3DPrintDemon/Demon_Klipper_Essentials_Unified #2: Calibrate_Z

Oxcart22, 2024-06-06:
I noticed that there is no ability to run Calibrate_Z during the print_start macros. Is there any specific reason for this? I usually have it run after z_tilt and bed meshing have completed and right before the print starts. Would this interfere with the other z offset functions that you have working?

3DPrintDemon, 2024-06-06:
> I noticed that there is no ability to run Calibrate_Z during the print_start macros. Is there any specific reason for this? I usually have it run after z_tilt and bed meshing have completed and [...]

Hi! If you mean to do so on a Sovol SV08 machine I really wouldn’t recommend it. I tried it many times on the machine I have here and it only gave bad results. 

I wouldn’t feel right encouraging people to use the stock system that way as it is now, it requires more work from Sovol to be accurate & reliable across a large sample range of machines. 
if I implemented it as default it will only result in damage to people’s machines & lots of angry issue posts being created here. That’s not good. 

If however you mean to use it on an accurate Voron tap system it is certainly possible to add Z calibrate at the end of the start macro if you wish, simply choose where you’d like it & paste it in. There’ll be no issue there. 

I might look into adding it as a default option for the next release or doing a mini update or something, I’d have to add it along with a strong warning not to use of SV08 machines until they improve their system.

3DPrintDemon, 2024-06-06:
> Thanks for the quick reply! That definitely makes sense. I'm using an Enderwire type build with a klicky probe, so I've had good experiences with calibrate_z in the past. That makes sense to want to avoid any bad outcomes with Sovol machines, but an implementation into this macro set would be very much appreciated. I would just be aware of how it might conflict with some of your other z offset macros such as anti-squish. I know there's a way to have them work together, but it might take some tweaking of your other macros.
> 
> The other thing I that I thought I would mention is that when using a docking probe there seems to be an excessive amount of docking and undocking the probe between each macro. i.e home; dock, z-tilt; undock, finish z-tilt; dock, bed mesh; undock etc. If there was a way to have these all happen in sequence without these extra motions it would speed up the start print considerably. I was using the print.cfg macros from Rootiest's github for a while, and his implementation didn't seem to behave the same way.
> 
> All that being said, this is an awesome project and I really appreciate you making this available to people. Seems to be working great so far!

That's cool, nice to have some feedback & I'm so glad you like the project! 
Yeah they have a similar system but not quite the same as the automatic Z Calibration plugin using a load cell. It's basically a Ballpark system at the moment.

The process for applying the offsets would be best done after any auto Z calibration so not to confuse anything I guess, not a big change, possibly a simple cut/paste job there.

Ah ok I get ya.  I use Klicky on my Voron 2.4 it only mounts the probe once only for the QGL, just wondering are you using Klicky for homing Z on the bed & not an end stop switch? Because the start macro simply calls the macros used by Klicky. The macros currently have no control over mounting or docking of the probe, so to keep the probe attached we'd need to modify the start macro's homing routine a little. Then it'd home, stay docked, do tilt adjust, meshing, auto Z & then dock. 

The only thing is for printers that need the probe docked to home Z, but I can do some thinking on that.

I'm happy to implement this for you as I think its a great to improve efficiency. I'll try to add it this coming weekend, I'm currently rebuilding my Voron & have a few print jobs too.

closed 2024-06-11 by 3DPrintDemon