3DPrintDemon/Demon_Klipper_Essentials_Unified #3: need help with the macros

NickDK1, 2024-07-12:
how does one change the heatsoak time on the sv08 nothing changes if i change the 300 in the sovol sv08 heatsoak cfg.
and how does one make the printer go in the far left courner when its finished printing amd print gets canceled like stock sv08

seenit783, 2024-07-12:
i was able to get the heatsoak macro kind of working but it was something i
found on internet. it kinda worked... had some issues.

NickDK1, 2024-07-12:
no matter what i change the value too when printing it does a 10m heatsoak

3DPrintDemon, 2024-07-12:
Hi you should not edit the macro files. All adjustments must be made in deomon_user_settings.cfg file, if you edit the macro files themselves you risk breaking functionally or worse, your printer.  

To change the heat soak timers you must edit the low & high temperature chamber timer variables in the demon_user_settings.cfg file. 

It’s down past half way after all the filament settings. 

Please watch the posted video on the main page for more info on this & other settings.

3DPrintDemon, 2024-07-12:
This may help. 
[image]

seenit783, 2024-07-12:
Oh I got however long I wanted but weird issues outside of heatsoak. Can't
remember. I'm on mainline klipper with pi4 and octopus pro amd haven't
tried that macro yet.

3DPrintDemon, 2024-07-12:
Im on the same basically, Voron 2.4 & the SV08 on BTT mainboard & Rpi4. No weird issues here. What were you experiencing exactly?

NickDK1, 2024-07-12:
no kamp
when printing it wasnt live adjudsting z after the height map
and super slowdown where it was suddently only was running 3mm/s

so i have gone bk to standard 
im apparently not able to set this up correctly

closed 2024-07-12 by NickDK1

3DPrintDemon, 2024-07-12:
> no kamp when printing it wasnt live adjudsting z after the height map and super slowdown where it was suddently only was running 3mm/s
> 
> so i have gone bk to standard im apparently not able to set this up correctly

Are you on a version of Klipper where their Adaptive Mesh is implemented?
Also have you activated `variable_adaptive_meshing: True` in the `demon_user_settings.cfg` file & followed the correct procure required for that system to work?

In the `Moonraker.conf` file, be sure the section below is present.
```
[file_manager]
enable_object_processing: True
```

Then activate `Label Objects` in your slicer?

TBH I have no idea why the actual print would slow down suddenly. There are no macros in this pack that would suddenly slow the printer in the middle of the print. There is a feed rate setting but is called one time only at the very start & that would mean the print would be that speed for the entire print unless changed.