3DPrintDemon/Demon_Klipper_Essentials_Unified #4: Parsing error's

royalfifty8, 2024-07-21:
Hello. I'm going through endless Klipper error's say "parsing error. I fex it the macro, restart firmware. Then same error message again but with a whole new list of lines to correct.

>

3DPrintDemon, 2024-07-21:
Hi there aren’t any errors in the macros. They’re used by many people on many different machines. Including my own machines. 
As you haven’t provided any examples of the errors you’re getting I can only assume it’s due to you not setting the required pins correctly or installing the additional prerequisite files for these macros to work. 
if you don’t do these things then you will have issues. 
but these thing you must do yourself as I can’t include them here, the way the licensing works.   

If you’ve edited lots of different macro files I’d recommend you uninstall the whole pack & reinstall it again & please try again. The documentation has also been recently updated.

royalfifty8, 2024-07-22:
Hey thanks for the quick response. I did have it working at first but now after reading the other issues, I believe the problem is that I edited the macros files not realizing that it messes up everything else. Going to just reinstall everything and try again. Really excited to use this so hopefully everything works after this. Appreciate the effort you've put into making this, the scope of work is mind boggling.

3DPrintDemon, 2024-07-22:
Thats ok, no worries. You're most welcome, I must say it has taken a considerable chunk of time to develop & test these macros from scratch. Hopefully you have a better experience second time round with your next install.

closed 2024-07-22 by 3DPrintDemon

royalfifty8, 2024-07-22:
I've reinstalled everything. Was not getting any errors until I went into demon_user_settings.cfg. I'm using an Ender 3 pro and I don't have bed fans. So I turned the option to [false]. After I clicked [save & restart], this Klipper error came up ---   " Option 'variable_pla_bed_fan_enable' in section 'gcode_macro _START_VARIABLES' is not a valid literal: malformed node or string: <ast.Name object at 0xffff851d23a0> ". I went back into the demon_user_settings. Reverted back to [True]. Restarted klipper and it went away but then [abs_fan_enable] decided to join the party. This also happens with - 
- variable_PETG_anti_squish
- TPU_anti_squish 
- default_bed_fan_enable
- variable_chamber_temp_wait
Once I reverted those back to [True]. Everything else seems to be working accordingly.

3DPrintDemon, 2024-07-22:
Thats really odd! I only have one machine with bed fans here & my others don't have them & I'm running those with the options you listed above disabled all using the same files version as available here. Same for the Anti_squish stuff. 

I am able to recreate your error message though by not entering the variable state correctly.

So if I enter `false` or `true` I will get:
```
Option 'variable_asa_bed_fan_enable' in section 'gcode_macro _START_VARIABLES' is not a valid literal: malformed node or string: <ast.Name object at 0x7fa7545c70>
```

However if I change that to `False` or `True` the system starts normally. This is a requirement of Klipper & its coding language, it has nothing to do with the actual macros themselves.

royalfifty8, 2024-07-22:
Well butter my butt and call me a biscuit! It worked! Thank you my good sir. Greatly appreciated and much respect for the prompt resolution.

3DPrintDemon, 2024-07-22:
You're welcome, yeah Klipper can be a tricky & unforgiving thing, it's so easy to get errors & mistakes, getting the capitalisation wrong of even the correct letter used can cause the entire system to fail. I'm just so glad we got you up & running! Happy printing!