3DPrintDemon/Demon_Klipper_Essentials_Unified #5: Smart Park Issue

halszkaraptor, 2024-07-27:
Thanks so much for all your work with these macros; they're great and reduce the amount of work required so much. I am having an odd issue with Smart Park though. This is on a Sovol SV08. I have the options configured right now to also use the adaptive mesh. For some reason, after doing the adaptive mesh. It does a short bust of movement towards what looks like the smart park position but stops and sits near where the mesh ended and tries to start the purge and print from there. In the console it outputs that Smart Park is executing and lists a valid location. The printer just never makes the move there.  I wrote a quick short custom gcode and put a G90 G1 X0 Y0 before the smart park and it worked correctly. I went through all the install steps several times and I really don't think I missed anything, or at least nothing I could find. I am really not sure if it is the macros, me or Smart Park. If you had any assistance you could provide though, I would greatly appreciate it. If there is any additional information I can provide, please let me know.

3DPrintDemon, 2024-07-28:
Hi thanks for posting, so glad you like the macros.
That does sound like s very strange issue. It could be something with the Kamp files or even something confusing the system from your sliced files. 
I just testing the adaptive meshing option again here using a sliced file from my copy of Orca Slicer & it worked perfectly I have to say.

The proper start process should be.....
After the first part of the start macro the printer will QGL, Z home, then build the adaptive mesh. 

It'll then clean the nozzle & move to the park location at the set height where it'll heat to printing temp for the purge line.
After its reached temperature it'll move over to the purge location at the set layer height & slowly purge. 

Then it'll start the print.

If you feel your Kamp files might be the issue you could try downloading a new set & replacing the ones on the printer, or using VSCode to compare the files for differences that could be causing your issue.

Also please check you have the slicer set up for use with adaptive meshing, plus you have the correctly defined `[file_manager]` section defined in your `Moonraker.conf` file, as well as `[exclude_object]` in your `printer.cfg`

Here's a link the relevant section in the Readme. Im sure you've done it but already but I'll add incase, & if anyone else reads this & requires the link.
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified?tab=readme-ov-file#setting-up-klippers-adaptive-mesh-option

halszkaraptor, 2024-07-29:
So I have no idea what happened but I ran a print with your macros again earlier and everything worked fine. Very very weird. Fingers crossed it was just a weird fluke. I did want to mention instead of KAMP Lite, I am just using regular Kamp with only Line Purge and Smart Park uncommented. I did that just in case they update their repo, moonraker update manager can install the updates without waiting for updates to the fork. 

Also, unrelated to this, I wanted to thank you for posting about the Klipper Expander board on the SV08 Facebook group. I had no idea that even existed and it makes adding things so much easier. I bought one right after I saw your post about it and already have it installed on the printer with extra fans and in the future will add a thermistor for chamber temp. 

All good to close this issue, thanks.

halszkaraptor, 2024-07-29:
Slipped my mind to also mention just in case, I have the nozzle cleaning turned off. Before everything worked correctly, I thought maybe this was the cause. It seems not though. In any case, if I have this pop back up, I will let you know.

halszkaraptor, 2024-07-29:
Spoke too soon. I just ran another print and it did it again. After meshing, did not move to the Smart Park position. I think I might go into the start print and add something to move the print head to the origin before smart park and see what happens.

halszkaraptor, 2024-07-29:
I went ahead and tried that by adding G90 then G1 X0 Y0 Z0.8 F9000 before SMART_PARK. It moved to the origin and just stayed in that area. Did not move to the smart park position and did the print near the origin instead of the center of the bed as I had sliced it.

3DPrintDemon, 2024-07-29:
Ok so this definitely sounds like a machine specific issue here with that intermittent behaviour. 

As an extreme test you can flash a fresh SV08 image to an SD card, backup your Demon folder & other required cfg files & then power down & safely remove the printer's EMMC module & insert the new system on SD card into the Printer's internal SD card slot & boot.

Copy back the Demon folder, prerequisites, & my_macros.cfg files to this new system & test. Try to only copy over what you need to & not everything.

That way we eliminate the current system. That then leaves your system's files & your slicer files.

3DPrintDemon, 2024-07-29:
Glad you got a Klipper expander in there too now! they'll be vital for SV08 users I think.

halszkaraptor, 2024-07-30:
Thank you for the advice. I will report back in a few days most likely. I'd like to muck around a bit more before I try that. I did so much flashing getting the printer to mainline that I'm burned out for right now 🤣 I didn't realize the stock 8GB emmc didn't work with the BTT CB1 image. Then I spent a bunch of time getting the 8GB emmc to work (because it bugged me that it did not and I felt compelled as a challenge) learning about device trees and such. Got that working then after looking through the device tree more, decided I just wanted to use the Sovol image since it is their board and they had made some changes in the device tree that I felt should be kept. I proceeded to uninstall their klipper, create a new user and re-install klipper. Anyway, after I mess around with gcodes and stuff a bit more, if nothing works, I will try the flash. Such a frustrating issue. I've been using KAMP for a while and never had an issue before and the first GCode related issue I have been at a loss to solve. I don't see why KAMP isn't working with a simple, bare bones start GCode. Sorry that was a bit long, thanks for your help so far.

halszkaraptor, 2024-07-30:
One additional small piece of information of wanted to mention in my testing this evening is I noticed in addition to the console output the correct smart park location, the mainsail UI also display the correct coordinates but that is not where the printer is located. So frustrating. Everything is running so smoothly with the exception of this one thing. I think tomorrow I might just flash the sd card, I'm about of of things to try.

3DPrintDemon, 2024-07-30:
Wow yeah sounds like you've made quite a lot of system based modifications there. I'd definitely try the system back on the stock Sovol Image with an SD card, you could try your files & even a freshly downloaded set of macros too to see how it goes. A bit annoying I know but I always say if you can't find what it is the best thing to do is find out what it isn't & go from there. 

At the moment I've got this current version of these macros running on 3 machines here, a Voron 2.4, Sv08, & a little SV07 which is basically stock. The SV08 ran these file perfectly fine previously on the stock system & now also on my upgraded system with new BTT hardware & true Main Branch Klipper (not Mainline), so no file or system editing involved.

All of them have worked with the macro files & the Kamp files too. So it's really odd that your system seems to struggle with this part of the process. The only differences that could make a potential problem are maybe the edits that have been made to the system. Also tbh it might be worth raising an issue on the original Kamp repo too. See if someone there has ever experienced this type of thing with their files.

halszkaraptor, 2024-07-30:
Welp, flashed it, installed the bare minimum, was working. Then I installed a few other pieces including the macros and Shake/Tune. Then it started acting up again. The problematic behavior seems to be it is having a bust of movement towards a location but not completing the movement causing everything to be off. It is now occasionally doing this before the adaptive mesh and even before a QGL. I have been using the tuned driver value from the SV08 mainline github. I am going to back those out and see if that fixes it. I have no idea what would be causing this other than that.

halszkaraptor, 2024-07-30:
I might end up making a post in the discord if I can't sort it out, this is so weird and no idea what up really.

halszkaraptor, 2024-07-31:
Well I think I finally nailed down the cause. I was looking at the smart_park macro and saw it is using what I think is 60% of the max velocity (420 when max is 700). I thought maybe this was just too fast and changed it to 42% to be around 300 and did the same for line_purge. It seemed to work after a few tests. I switched back to my imaged emmc before I flashed it again then put this change in and I haven't had it fail to smart park since I made this change. I might just add a travel speed variable. I'm not sure why the higher speed was causing an issue but hopefully this doesn't come back up.

3DPrintDemon, 2024-07-31:
Hi! Nice work there finding the problem on your machine. 
At this point it pains me to say it’s a shame you didn’t use my KAMP_LiTE fork from the start as I have halved the travel speed there on both their purge & park macros. So I’m guessing this issue probably wouldn’t have been a problem for you as you would’ve never experienced it, if it was indeed down to travel speeds.

halszkaraptor, 2024-07-31:
Oh gosh, that's so ironic, but for sure how things are for me sometimes with stuff just randomly going wrong 🤣 thanks so much for all your feedback as I was working through this. I really appreciate it. Finally have the sv08 just about where I want it. Only things left to do are install the 32gb emmc when it arrives and switch out the inductive probe.

3DPrintDemon, 2024-08-01:
Yeah the way life is sometimes I guess. At least it's hopefully sorted now on your system! You're most welcome.
Sounds like some choice upgrades there, the stock storage is a little small & fills up quickly.
Also be sure to update your probing & meshing settings for improved accuracy.

IMPROVE YOUR PROBING & MESHING!!
Amend your stock settings with these values, they're not complete sections in some cases so DO NOT just copy/paste over the top of your current sections! Double check the values & don't delete anything you shouldn't!

 
```
[printer] # THIS FIRST SECTION IS OPTIONAL TO SLOW THE MACHINE DOWN

kinematics: corexy

max_velocity: 700

max_accel: 8200

max_accel_to_decel: 4500

max_z_velocity: 15

max_z_accel: 500

square_corner_velocity: 5.0
```
 
```
[probe]

speed: 5

samples: 3

samples_result: median

sample_retract_dist: 5.0

samples_tolerance: 0.0125

samples_tolerance_retries: 10
```
 
```
[bed_mesh]

speed: 350

horizontal_move_z: 5

 

probe_count: 9,9

algorithm: bicubic

bicubic_tension: 0.4

split_delta_z:0.0125

mesh_pps: 3,3

# move_check_distance: 3

adaptive_margin: 5
```
 
```
[quad_gantry_level]

speed: 350

horizontal_move_z: 5

retry_tolerance: 0.0125

retries: 5

max_adjust: 10
```

halszkaraptor, 2024-08-02:
Yep, luckily I had already done that. When I went to mainline, the repo working on it had values for changing the probing so I started directly with those. The sample and retry tolerance are both at .0075 as well slower speeds. Hopefully I won't have to use it for too much longer though, I have the Eddy probe all ready to install; just need some time to do it. The inductive probe is driving me nuts, I'm very used to the biqu microprobe I had before. I saw some mounts for it but I don't really feel like messing around with splitting the cable cable and adding a resistor. Thanks again!

3DPrintDemon, 2024-08-11:
Hi sorry for the delay in replying. Yeah some Sv08 probes are not quite accurate enough sadly. The Eddy does look cool. I have one here too. Like you I need time to play with it! The Sv08 is a really nice machine once issues have been corrected.

closed 2024-08-11 by 3DPrintDemon