3DPrintDemon/Demon_Klipper_Essentials_Unified #6: MACHINE_DEFINE option  is missing in demon_user_settings_v2.8.cfg

FoldZero, 2024-08-01:
MACHINE_DEFINE option  is missing in demon_user_settings_v2.8.cfg

3DPrintDemon, 2024-08-01:
Awesome! Thank you for your contribution, I appreciate it, I'll get onto these ASAP!

Although the MACHINE_LEVEL_HOT/COLD macros should home the printer if its not currently active.

FoldZero, 2024-08-01:
if your interested ( i haven't used git much) but I've attempted to make the changes for my own version
This [branch is [3 commits ahead of]  ](https://github.com/FoldZero/Demon_Klipper_2.8_Dev/compare/3DPrintDemon%3ADemon_Klipper_Essentials_Unified%3Amain...main )contains the changes

also i couldnt find the [gcode_macro _HEAT_WAIT] anywhere, im stumped. its used in the bed leveling if there isnt a sensor.

3DPrintDemon, 2024-08-11:
Hi sorry for the delay in replying. I'll take a look your folk & will modify things here as building options for filaments would be a good thing I think!

The HEAT_WAIT macro is located in my fork of these macros which should be included.
https://github.com/3DPrintDemon/Non_Blocking_Wait/releases/tag/Heat_Soak_Timers_V1.0

As per the file instructions. You're probably right, they are still in there from older versions. So will be easy to strip those back. No point on having it all duplicated I guess.

I'll see what can be done about 0º temp settings.

closed 2024-09-10 by 3DPrintDemon

OoSTARTERoO, 2024-12-13:
I suggest that _HEAT_WAIT should be included in this Unified macro pack
It gets called, but not defined on any of the folder

3DPrintDemon, 2024-12-13:
> I suggest that _HEAT_WAIT should be included in this Unified macro pack It gets called, but not defined on any of the folder

Hi the _HEAT_WAIT macros are a forked repo & used under GPL licenseing so cannot be directly included in this pack, sorry. For any future reference if no folder location is specified for a file then it means it can go in your main config folder on your printer & simply directly included in the printer.cfg