3DPrintDemon/Demon_Klipper_Essentials_Unified #9: 2 Problems i can't solve myself.

Mad-Virus, 2024-08-11:
Hi,

i have 2 problems that i can't figure out. I am an absolute Klipper Newb.

First is, the LED Stuff. Because i changed my SV08 to Mainline the SV08 RGB/Heatsoak Solution wont work for me with the original config Snippet. pin:EXP1_6 throws an Error like EXP1_6 is not an MCU Pin. I checked the Sovol original printer.cfg and found aliases there so EXP1_6=PC10. I Changed it in the Snipped for the printer config to PC10 and also in RGB_LEDs.cfg. Will this work like it should?

Seccond, the mainsail.cfg Usage section is unclear for me. I created a new file my_macros.cfg Copied the _CLIENT_VARIABLE Stuff, uncommented it, made my changes. All fine until here. Also the mainsail guides i found didnt helped me. Mainsail text says to copy it to the printer.cfg and put it under [include mainsail.cfg], you also talk about using [include mainsail.cfg] but put it into another file. Do i have to put also [include mainsail.cfg] on top of the file or do i have to make a [include my_macros] to the printer.cfg? it feels like something has to be done to call it for use. I dont want to have it in the printer.cfg file.

Greetings
Alex

3DPrintDemon, 2024-08-12:
Hi I can't speak for any setups on different Klipper Repos but my understanding is that it should probably work. You must define the led pin once in the correct section. To know if it works correctly your screen & screen button will change colour with any STATUS_xxxx macro used.

Place in your printer.cfg file...
```
[include mainsail.cfg] 
[include my_macros.cfg]
```
...be sure to check your file spellings for the above.

Then in your `my_macros.cfg` file paste in the `_CLIENT_VARIABLE` macro stuff & then the Klipperscreen LOAD/UNLOAD macros below it. 

You should be all set!

Mad-Virus, 2024-08-12:
Thanks for reply. 

So it sounds like the LED is from original Sovol LCD or  Display? I'am using the BTT 5hdmi v1.2. Is there a way to show it there? ATM it looks like i am using a dummy pin.

I switched to mainline via the Rappetor Guide for sv08. So i used the config from there.

3DPrintDemon, 2024-08-19:
> Thanks for reply.
> 
> So it sounds like the LED is from original Sovol LCD or Display? I'am using the BTT 5hdmi v1.2. Is there a way to show it there? ATM it looks like i am using a dummy pin.
> 
> I switched to mainline via the Rappetor Guide for sv08. So i used the config from there.

The dummy pin is what you need unless you install your own neopixel LEDs.

3DPrintDemon, 2024-08-19:
> So another Problem showed up. Klipperscreen LOAD/UNLOAD Macros. I dont know this should be the way but you have to heatup the nozzle your own to use that buttons because the Macro only tries to load or unload also with a cold nozzle.

This is not an issue. This is how the macro is supposed to be as it’s meant for use while printing so the nozzle already up to temp & as there’s lots of different temperature filaments available so you are best to set it for what you are using. 

I have been considering writing my own as these are indeed a little basic.

3DPrintDemon, 2024-08-19:
You can use any LOAD & UNLOAD macro you like. They're not integrated into the macro pack & are only called by the user with buttons in mainsail or on Klipperscreen.
The LOAD_CLEAN & UNLOAD_CLEAN macros are integrated.

closed 2024-08-21 by 3DPrintDemon