3DPrintDemon/Demon_Klipper_Essentials_Unified #11: error on start print

igeighty, 2024-08-19:
hello, when I start a print using your macros, I get a following error. I'm sorry I don't know what to look for here.. 

Error evaluating 'gcode_macro _START_WAIT_AND_TIMER_HANDLING:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'BED'

let me know if you need any of my files .

thankyou...

igeighty, 2024-08-19:
user error I would say... the filament type was HS-PLA+. when I set it to PLA+ in orca. print starts normally,
Very sorry, it was on the second read through your instructions that I found this. sorry to waste your time.

brilliant work on the code btw,.

closed 2024-08-19 by igeighty

3DPrintDemon, 2024-08-19:
Hi! Thanks for posting, not wasting time at all! This is actually an issue you found!
As when the filament type is not recognised by macros as the main 5 sorts it should fall back to read the bed temperature for the file & make a decision based on that, but looks like that’s not working right now. So I’ll get to fixing that! 
Thanks again for finding this, appreciate it. So glad you like the macros.

igeighty, 2024-08-19:
[image]
Thanks so much for writing these macros,. I had just finished building the Voron 2.4 and went straight to your repository.
its just amazing, and combined with your macro set. it is just even more so....

its my OCD that makes me label things as such, in the image above, under 'TYPE' I had entered HS-PLA+ in the text field, and this was the cause of the error, I changed it to PLA+ and it works perfect. I changed it back to HS-PLA+ and reproduced the error..

one question if I may though. am I able to disable the 10 minute heat soak that occurs when starting a print ?

thanks again. I appreciate your work.

3DPrintDemon, 2024-08-19:
That’s ok, yeah these were written for my Voron 2.4 also, but I wanted them to be usable on my other machines as well. So glad you’re enjoying my work. 

Ah yes a “Type” set as PLA, PLA+, PETG, TPU, FLEX, ABS, ASA will work ok for handling via filament type, then anything outside these should be handled via bed temperature. It’s the last part here that’s broken right now so it throws an error. Just needs a little update. 

You can edit timer values in the user settings file & you can even totally skip the timer’s altogether by setting the “start_my_print_already” variable to true.

Then if you need a sinlge heat soak at the start of printing set your bed temp & start a manual timer. 

However I recommend installing a chamber fan with thermistor. Then the printer will automatically run a temperature wait heat soak if one is needed or not. The time of the heat soak is automated too. 

It will also start the chamber fan & cool the chamber of it gets too hot while printing (standard klipper feature) & will also modify the fan target temp once the print is completed so the chamber is cooled post print too. 

Settings for these in user settings file.

igeighty, 2024-08-19:
brilliant mate, thankyou so much for the explanation.. love the user settings file. I have a lot of reading to do.

hope you have a good one :)

3DPrintDemon, 2024-08-19:
You’re welcome, happy printing!