3DPrintDemon/Demon_Klipper_Essentials_Unified #12: not a problem, but a question

igeighty, 2024-08-21:
Hi 3DPrintDemon,

I was wondering if you knew if anything in your scripts is telling the machine to extrude at 96%. I was chasing under extrusion problem and noticed in KlipperScreen under fin tuning that he extrusion was set at %96, I double checked slicer and its not telling it so I wondered if it was DKE playing a role.

again sorry to create a ticket, just something I can't work out myself. 
[image]

3DPrintDemon, 2024-08-21:
There’s a very simple solution. Did you watch the setting’s tutorial video or look in the user settings files for filament settings? You’ll find the answer there.

closed 2024-08-21 by 3DPrintDemon

igeighty, 2024-08-21:
thankyou :)