3DPrintDemon/Demon_Klipper_Essentials_Unified #14: Adaptive_Meshing

aquari86, 2024-09-02:
Hello, I always get the error message Error evaluating 'gcode_macro PRINT_START:gcode': gcode.CommandError: This error is caused by the sliced ​​file not having LABEL_OBJECT enabled or by your slicer's start Gcode not being setup correct! Please disable Adaptive_Meshing in the demon_user_settings.cfg or re-slice the file with it enabled or your start Gcode corrected and restart the print!

I followed the instructions and activated "Label Objects" in the Orcal slicer, inserted [exclude_object] in the printer.cfg and inserted
 "[file_manager] enable_object_processing: true" 
in the moonraker.conf, but I still get the error message. Without Adaptive_Meshing I only get worse results. What can I do?

mojamuto, 2024-09-03:
[image]

are you sure you have the two orcaslicer checkboxes checked (tag objects and exclude objects) in the output gcode?

aquari86, 2024-09-03:
Yes I have it activated, if I deactivate adaptive meshing, the error message "Error on 'M221 S': unable to parse" appears

3DPrintDemon, 2024-09-03:
Hi. If you have definitely followed the instructions & verified your system is ready for & can use adaptive meshing then it points to an issue with your slicer’s start Gcode. Please take a screenshot of it & post it. 

An error with M221 is very strange as that is a flow rate setting. Please verify all filament flow settings are correctly set in the demon_user_settings file.

aquari86, 2024-09-05:
sorry, I'm a bit embarrassed now but I forgot to rename the folder, now everything works except that I have the problem that when I print several things, each thing is printed with a different z-offset, I printed several squares and each had a different z-offset, then I printed a large square that went over the whole area, the z-offset was OK, what could be wrong with that?

3DPrintDemon, 2024-09-06:
Hi. No worries. Glad you got it working. Strange regarding the different Z offsets. There’s nothing in the macro pack that will do that so I would assume it’s something else on your system, a problem with your sliced file or simply an inaccurate bed mesh maybe.

aquari86, 2024-09-07:
I think I have found the reason, the printer always lost z-offset, even while it was printing it got worse layer by layer, I think it was due to the mainline settings of Rappetor, run_current and hold_current were changed, which caused the printer to lose height, I changed all the values ​​of run_current and hold_current to the original values ​​and it works, thank you for your help and your work on the macros

3DPrintDemon, 2024-09-10:
Hey great you found the cause! Please be sure to pass this info on to the relevant people who made your printer.cfg file, they should know about the problem you've had with your Z height & how to fix it.

You're most welcome, glad you like the macros! Happy Printing.

closed 2024-09-10 by 3DPrintDemon