3DPrintDemon/Demon_Klipper_Essentials_Unified #16: Part Cooling Fan Staying On After Print

halszkaraptor, 2024-09-06:
After a print ends, the part cooling fan is staying on. I have reviewed everything and can't find anything that I missed or did not configure. Not sure if it is me or the macro. This is on an SV08 and part colling fans are configured as below. Shutdown relay is false, neopixel is true, chamber heater is false and nevermore is false. 

```
[multi_pin fan_pins]
pins: extra_mcu:PA7, extra_mcu:PB1

[fan]
pin:multi_pin: fan_pins
max_power: 1.0
```

3DPrintDemon, 2024-09-06:
Sounds like the orca slicer profile maybe still has additional/secondary parts fan enabled. 

Please verify this option is off.

halszkaraptor, 2024-09-22:
My apologies, I got extremely side tracked in other things and it slipped my mind I opened this issue. Thanks for replying, I will check that. I've been using Cura for my PLA prints and I never thought to check if it is staying on for those because I've been doing mostly ABS lately (using Orca for those).

3DPrintDemon, 2024-09-23:
Thats ok. Also check if the Sovol macros are correctly disabled.
There is a M107 command in the end code so it should (& does) kill the parts fan named [fan]

halszkaraptor, 2024-10-23:
I've been taking a break from 3d printing so I'm gonna say go ahead and close this issue. I'm sure the issue is what you suggested, thank you 🙂

3DPrintDemon, 2024-10-23:
Ok no worries. It’s probably actually due to having the additional fans enabled in orca slicer.  Will close this issue.

closed 2024-10-23 by 3DPrintDemon