3DPrintDemon/Demon_Klipper_Essentials_Unified #23: Pressure Advanced Test Mode.

Mad-Virus, 2024-09-18:
So after i updated to the Fixt version i tryt again to use the PA Test mode. 
Is it me that i dont know what to do with it? It now shows me the text it showed me also bevore in the Error text and than do nothing.
What does it want from me? 

Greetings
Alex

3DPrintDemon, 2024-09-18:
Hi. It automates the setup for Klipper’s pressure advance test. Not used with the Orca PA test. 

Please see here. 
https://www.klipper3d.org/Pressure_Advance.html

Mad-Virus, 2024-09-18:
Yeah i found that and do it by hand. 
So it only set the variables you find in the doc ? Cant check it atm because of printing the square tower.

Mad-Virus, 2024-09-18:
You realy should have a Guide to the Functions of your Macros. But i Know, Time is short for everyone this times.

But finaly. After all that work since July this small little B..... prints me a square tower. I realy cant belive it. Hope there will be no other problems to solve in next time.

Thanks :)

3DPrintDemon, 2024-09-23:
Yes I agree! When I have some time I'd like to make a guide to the macro functions.

Happy printing, fingers crossed for no more issues.

closed 2024-09-23 by 3DPrintDemon