3DPrintDemon/Demon_Klipper_Essentials_Unified #25: BTT EDYY

ahmedbebars, 2024-09-28:
Is it possible to run BTT-EDDY

3DPrintDemon, 2024-09-28:
Hi! Full & automatic Eddy probe integration is coming in a few days with the v2.9 version update!

just doing some testing etc now!  Keep an eye out, you won’t want to miss it! 😉👍

ahmedbebars, 2024-09-28:
Thank you for your interest and quick response.

ahmedbebars, 2024-09-30:
Thanks for the update 

There are some problems when starting printing, as shown in the pictures
[image]
[image]

3DPrintDemon, 2024-09-30:
Hi looks like you haven't install the timers correctly as that is the _HEAT_WAIT command. The SET_PIN command could well be related to that.

Please install the heat_wait timers here:
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/Prerequisites/Prerequisites.md

closed 2024-10-09 by 3DPrintDemon