3DPrintDemon/Demon_Klipper_Essentials_Unified #26: //Unknown command:"_DRIVER_DEFINER"

patofoto, 2024-09-30:
Have placed the _DRIVER_DEFINER line following your instructions. Have a Klicky Probe. Getting this message but printer homes properly.

[image]

[image]

Did I miss something?

3DPrintDemon, 2024-09-30:
so sorry, typo

change to 
```
_DRIVER_DEFINER
```

3DPrintDemon, 2024-09-30:
now corrected in documentation

patofoto, 2024-09-30:
Awesome. I should have spotted this. Duh. Thanks.

3DPrintDemon, 2024-09-30:
Thats ok me too! it was my mistake as was in the documentation. What happens when you write macro instructions past your bed time! hahaha

closed 2024-09-30 by patofoto