3DPrintDemon/Demon_Klipper_Essentials_Unified #27: Unknown command:"_Z_RAISE"

patofoto, 2024-09-30:
Getting this when pausing or canceling.

3DPrintDemon, 2024-09-30:
This was changed this update, in previous version (<2.8) `_Z_RAISE` was used in the `mainsail.cfg` `_CLIENT_VARIABLE` macro.

However this has now changed to the following.
[image]
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#mainsailcfg-usage

patofoto, 2024-09-30:
Oh, I missed this. Had changes in so many places that my ADHD side was a bit overwhelmed. Sorry. Wonder is there is a way to update things without going over the whole documentation. All my mistake.

3DPrintDemon, 2024-09-30:
That’s ok don’t worry. Yeah I’ll make a “Things you need to change if you’re updating” file. That should help I think. 😉👍