3DPrintDemon/Demon_Klipper_Essentials_Unified #28: help: cant home xyz axis

sightanignat, 2024-10-01:
when i press the home all axis in mainsail this error appears:
[image]

3DPrintDemon, 2024-10-01:
Ohhh that’s a strange one. That command is a basic I built in klipper command, it appears however it’s unsupported in your Klipper branch? Is it an old version maybe?

Are you able to update or research if that unknown command `SET_KINEMATIC_POSITION` is available or not for your branch?

if not I can help you modify the homing process so you use the macros.

chbartsch, 2024-10-01:
I just had the same problem and was missing

`[force_move]
enable_force_move: true`

in `printer.cfg`.

But now I'm getting "The value 'AES_SYSTEM_ENABLE_-_KLIPPER_CONTROLLED' is not valid for PIN".

3DPrintDemon, 2024-10-01:
> I just had the same problem and was missing
> 
> ```
> [force_move]
> enable_force_move: true
> ```
> 
> in `printer.cfg`.

Just looking at that thanks for the solution there. Will be sure to add that to documentation. It’s always the little stuff that pops up if you already have it defined in your file!

3DPrintDemon, 2024-10-01:
> I just had the same problem and was missing
> 
> ```
> [force_move]
> enable_force_move: true
> ```
> 
> in `printer.cfg`.

Also, sorry for the edit on your comment. On my phone right now & it insists on zooming right in so clicked the wrong ##%^ ing button lol 🤪

3DPrintDemon, 2024-10-01:
Documentation updated to include the above info. 😎👍

sightanignat, 2024-10-01:
just as i finished downgrading to demon 2.7 and run smoothly hahaha. I guess gonna try 2.9 it now. thanks alot guys

sightanignat, 2024-10-01:
now the z axis moved up. and X axis homed but Y axis didnt move. new problem, 
[image]

3DPrintDemon, 2024-10-01:
ahhh ok looks like you dont uart enabled drivers. damn.

Ok...

Open the homing contol file. 

Comment out lines 125 down to 163 & paste this in underneath the commented out section.

Its the same macros for homing but without the change of drive current for UART drivers.

NOTE: This is untested but "should" be fine as the homing commands are exactly the same but without the power adjustments. BUT please do cover the emergency stop button just incase something weird happens.

If you wanted you can delete the G4 lines too, as you maybe don't need those.

```
[gcode_macro _HOME_X]
gcode:
  {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}

    G28 X
    G91
    G0 X-{start_vars.axis_backoff_distance|int} F{start_vars.axis_backoff_speed|int * 60}

    G4 P{start_vars.axis_register_clear_wait|float * 1000}

[gcode_macro _HOME_Y]
gcode:
  {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}

    G28 Y
    G91
    G0 Y-{start_vars.axis_backoff_distance|int} F{start_vars.axis_backoff_speed|int * 60}

    G4 P{start_vars.axis_register_clear_wait|float * 1000}
```

sightanignat, 2024-10-01:
just comment out 125 - 163 and paste the new _home_x and Y gcode. one more problem. 
[image]

3DPrintDemon, 2024-10-01:
fix that with this for now, I'll sort something out for that.
[gcode_macro _HOME_X]
gcode:
  {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}

    G28 X
    G91
    G0 X{start_vars.axis_backoff_distance|int} F{start_vars.axis_backoff_speed|int * 60}

    G4 P{start_vars.axis_register_clear_wait|float * 1000}

[gcode_macro _HOME_Y]
gcode:
  {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}

    G28 Y
    G91
    G0 Y{start_vars.axis_backoff_distance|int} F{start_vars.axis_backoff_speed|int * 60}

    G4 P{start_vars.axis_register_clear_wait|float * 1000}

sightanignat, 2024-10-01:
its fix now but every after homed axis theres a pause. but its fine. and one more, how can i edit my  x/y offset. my probe point  is a little off when it homes.

3DPrintDemon, 2024-10-01:
you can delete those G4 commands at the end. That will eliminate the wait. 

you need a specific probe point? the current probe point is simply 50% of you X & Y axes

sightanignat, 2024-10-01:
my bed is 300x300 and the probe  X offset is -31.80 and y is -40.50 so i need the probe to be at x181.80 , y190.50  when i homes.

3DPrintDemon, 2024-10-01:
ok got you covered, am rebuilding the homing macros for you with extra options. nearly ready to test now

ahmedbebars, 2024-10-01:
[image]

When he returns home, z does not move
and is supported tmc2240

3DPrintDemon, 2024-10-01:
> ![3](https://private-user-images.githubusercontent.com/35563932/372450990-438c1348-b67d-4b98-9791-9e34db7d82b6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc3ODM2NzksIm5iZiI6MTcyNzc4MzM3OSwicGF0aCI6Ii8zNTU2MzkzMi8zNzI0NTA5OTAtNDM4YzEzNDgtYjY3ZC00Yjk4LTk3OTEtOWUzNGRiN2Q4MmI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAxVDExNDkzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0NDU1OWY0NTFmYjVhOTc5YmE4NDU1Y2VlZmU2YmQ0YTBjY2U5ZDUwMjNjYTg0Y2UxNjBlYTEwNjliYTExODkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gBqfqwmDLAGm16ZO3c7ULQBzDUc5rszWF60YEdLrf90)
> 
> When he returns home, z does not move and is supported tmc2240

check your user settings file line 460
```
variable_aes_system: False                       # Do you have the Demon Auto Emergency Stop System True/False
```
should be false, updated that this morning

3DPrintDemon, 2024-10-01:
> my bed is 300x300, the probe X offset is -31.80 and y is -40.50. so i need the probe to be at x181.80 , y190.50 when it homes. and i have to limit the y max position to 294. the bed is bumping the belt tensioner.

Ok new files are uploaded! 

Please either download the demon_user_settings.cfg again or simply paste these at line 293 inside the DEMON HOMING CONTROL SETTINGS section.

```
variable_set_probe_point_default: True           # Set False if you want a custom probing point for your bltouch/eddy probe
variable_set_probe_custom_x: 100                 # Set custom X probe position for your bltouch/eddy probe
variable_set_probe_custom_y: 100                 # Set custom Y probe position for your bltouch/eddy probe
```
Then change `variable_set_probe_point_default:` to `False` & set your desired probing XY points.

The printer "should" now also see you have no supported  UART drivers & will automatically take into account your homing direction when performing axis backoff moves.

Please download & install:

[demon_homing_control_v1.1.cfg](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_homing_control_v1.1.cfg)

[demon_core_assets_v1.3.cfg](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_core_assets_v1.3.cfg)

BE SURE TO:

change the file extension name of the file `demon_homing_control_v1.0.cfg` to....
```
demon_homing_control_v1.0.txt
```
This will remove it from your system, don't delete it just yet as you have edited your one to be working for you. Delete after new files confirmed working.

3DPrintDemon, 2024-10-01:
ok damn loaded a test file with macro commented out for another machine.

go into the file & comment in the homing_override macro

Sorry too many file versions here right now!

sightanignat, 2024-10-01:
homing working now ,thanks alot for the help and effort, now im going to test all the macros.

3DPrintDemon, 2024-10-01:
> homing working now ,thanks alot for the help and effort, now im going to test all the macros.

FANTASTIC!!!! Yes!

You're most welcome! I'm just glad I had the time today to do the rewrites for you!

chbartsch, 2024-10-02:
> Just looking at that thanks for the solution there. Will be sure to add that to documentation. It’s always the little stuff that pops up if you already have it defined in your file!

It was in the Sovol macro file (SV08) which isn't used anymore. ;)

3DPrintDemon, 2024-10-02:
> > Just looking at that thanks for the solution there. Will be sure to add that to documentation. It’s always the little stuff that pops up if you already have it defined in your file!
> 
> It was in the Sovol macro file (SV08) which isn't used anymore. ;)

Yeah thats a really odd place to have it tbh, should be in the printer.cfg as standard. But hey-ho...

Aircoholic, 2024-10-16:
Hi, I'm facing a situation similar enough to tag along this issu, but different enough that I'm not sure if the above described solution would work for me. Before I experiment any further, I thought I'd ask.
I have a Anycubic Vyper running Klipper on a Raspberry Pi - if you need any other info, I'm happy to provide that.
When I'm homin the axes, Z lifts, X and Y home alright, and then I get the following error message, while Z does not move down.

[image]

Aircoholic, 2024-10-16:
> Hi there. The system is seeing you have a board pin set as an endstop switch so you need to go into the demon user settings file & find the homing control section. There you need to set the position of your endstop switch as the default 8888 values are meant to do exactly what you are experiencing & prevent homing of a system where these values have not been defined by the user.
> 
> Hope that helps.

Hi, thanks for the quick reply! It does help, kinda... 
In the Demon user settings file, I find the "VORON STYLE Z ENDSTOP SWITCH OPTIONS", which is the only place where I also find the 8888 entry. 

> variable_z_endstop_loaction_x: 8888   
> variable_z_endstop_loaction_y: 8888 

Now, I'm a noob when it comes to Klipper and have a little experience with 3D printers, and am therefore not sure what value to put in here. Any help or pointers in the right direction are much appreciated!

3DPrintDemon, 2024-10-16:
I just realised you have the Vyper, I’m assuming that has the stock strain gauge still installed. In this case you can simply set those values to the centre of your bed, you choose the X & Y position to use. 
However I have not tested these macros on this model printer so please be extremely careful when homing for the first time. Make sure your probe endstop is being recognised. Do not home directly on the bed first. Raise the printer up high & place something under the nozzle to trigger the sensor before it goes near the bed so you have time to hit the emergency stop button if needed. Also please be careful when first levelling & meshing.

Aircoholic, 2024-10-17:
After following your instructions, I nervously ran the homing procedure, successfully! And running a bed mesh routine was really quite some excitement, sitting by the printer, checking the probe doesn't crash into the bed - which it didn't :-D
I even ran a test print successfully and was super happy everything worked, adaptive meshing, purging, printing. However, now after that, there seems to be a bunch of issues.
After the print, I made some adjustments to the Z-Offset, saved config, rebooted and when I press the macro "ready PLA", which worked before, I now get the following messages:
[image]

My printer doesn't have a chamber and I'm not sure why this message now pops up. After the message, the printer goes into parking position, cooling down bed and extruder.
When I press the "ready PLA" macro again, I get this:
[image]

I've rebooted the printer and the Raspberry Pi several times, but I can't get any further than this.

3DPrintDemon, 2024-10-17:
The wording probably could be better on that chamber timer message. It’s just a heat soak timer to make sure your printer & more importantly your probe, as it’s most likely induction based is up to a stable temperature for better meshing/probing results. This feature can easily be turned off I. The user settings file of in the print start quick settings before you print. 

The other message tells you why the operation was denied. Your machine is not in standby, it is a safety thing that stops users running this directly after a print incase there is a model or something on the bed. 

Simply clear the completed, cancelled, paused, printing status of your machine back to standby and it will work. It is intended function.

Aircoholic, 2024-10-22:
I've got it all working now. Well most of it anyway... the z-offset value doesn't get saved after calibrating (hot or cold), but adding the value to Orca slicer did the trick - not ideal, I know, but it works.
And some features I only figured out after finding my way around Klipper a bit more.
Thank you for your support! My printer is finally printing again and better than it ever has!

Aircoholic, 2024-10-23:
This is probably getting wildly off-topic, but one question came up, after printing the Califlower calibration tool.
After measuring, I'm told my skew needs correcting and I should: "Add the following to the start of your start G-code macro, typically PRINT_START or START_PRINT found in printer.cfg or macros.cfg"
How does this translate to the Demon Essential Macros? I don't just want to add the skew correction somewhere, where it might cause issues, but where, in which file, do I add the extra line exactly?

3DPrintDemon, 2024-10-24:
> This is probably getting wildly off-topic, but one question came up, after printing the [Califlower calibration tool](https://vector3d.shop/pages/califlower-calibration-calculator-mk2). After [...]

Hi you should be able to add it to the very end of the PRINT_START macro after all other setup is done.
Note the caveats section here. 
https://www.klipper3d.org/Skew_Correction.html#caveats

3DPrintDemon, 2024-11-02:
> This might be super obvious, but I can't seem to find the PRINT_START macro - where exactly do I find it? It's probably not in the demon_print_start_end_v2.9.cfg file, correct?

Hi the PRINT_START macro is indeed in the demon_print_start_end_v2.9.cfg file, it’s the first macro in the file & is at the very top, it always has been. New version out now v2.9.1, just small updates mostly

Aircoholic, 2024-11-02:
Ah, it is that part of that file, cool. I saw the warning message and thought, I don't really know what I'm doing, so I didn't change anything - good to know that my intuition wasn't that far off.
Thanks for the support, much appreciated!

closed 2024-11-04 by 3DPrintDemon

Aircoholic, 2024-11-05:
Turns out, I don't seem to know what I'm doing. 
When sending a print to Klipper, I get a "Move out of Range" Error. 
When I remove the Skew Correction from the Start_Print macro, it works without the error.