3DPrintDemon/Demon_Klipper_Essentials_Unified #29: Ztilt vs quad_gantry_level on CoreXY

mmrrxx00, 2024-10-06:
I have a modified ender 5 plus that is using the ZeroG Mercury One. So it is a corexy but still using Ztilt for the Z axis. I did not see a setting to be able to define the usage of ztilt vs quad_gantry_level when using corexy as the motion type.

3DPrintDemon, 2024-10-07:
That’s a very good point! I’ve got something written up & I’ll post an updated file later today for you. Will comment again here once uploaded to let you know.

3DPrintDemon, 2024-10-07:
Update applied. Please re-download files:

`demon_core_assets_v1.3.cfg`
`demon_z_calibration_v1.6.cfg`

Hopefully that should give you the levelling system you need. Please note it is untested as I do not have that configuration of machine here. To test you should be able to run the `MACHINE_LEVEL_COLD` macro, but cover the emergency stop button when you go, just incase. I don't see why you would need to tbh as it should Z tilt for you now. The edited macro worked as it used to on my true CoreXY machine.

mmrrxx00, 2024-10-07:
Running the level cold macro worked great. Thank you so much.

closed 2024-10-07 by mmrrxx00

3DPrintDemon, 2024-10-07:
Hey! Nice to hear that!! Fantastic stuff. Thanks for letting me know & requesting that feature for your machine! Happy printing!