3DPrintDemon/Demon_Klipper_Essentials_Unified #30: homing

aquari86, 2024-10-08:
can you maybe tell me how I can change the home order so that Y goes home first and then X, thanks

3DPrintDemon, 2024-10-08:
Ive updated the files for you to enable this feature.

Please download file `demon_homing_control_v1.2.cfg` & be sure to delete `demon_homing_control_v1.1.cfg` from your printer!!!
NOTE: You must do this, you can't have two versions of this file on your system.

Please re-download file `demon_core_assets_v1.3.cfg`

Then you can either re-download the `demon_user_settings_v2.9.cfg` file & enter your settings again....

Or.....

you can open the current file you have & at line 285 in the `DEMON HOMING CONTROL SETTINGS` section paste in the variable below.

```
variable_home_y_first: True                      # Set True to home the Y axis first. Set False to home X first
```

This will now allow to to choose which axis to home first.

Please be sure to cover the emergency stop button just incase you have any issues. Hopefully not though as it worked perfectly fine here on my coreXY machine.

3DPrintDemon, 2024-10-08:
Can you check your variables file & make sure the home_y_first: is set “True” & not “true”

aquari86, 2024-10-08:
You are the best, I have a point behind true, my mistake, thank you very much, for your work and help

3DPrintDemon, 2024-10-08:
Are you using an eddy probe?

aquari86, 2024-10-09:
yes exactly I use eddy, is that a problem

aquari86, 2024-10-09:
I have now found the macro of btt and included it but now comes the error : Error evaluating 'gcode_macro SET_Z_FROM_PROBE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'safe_z_home'

3DPrintDemon, 2024-10-09:
Please be sure to read the published documentation as the steps you’re trying to accomplish are there. 

Please see this section & image of setting up the Eddy file & macros correctly. 

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#btt-eddy-probe

aquari86, 2024-10-09:
I did it, it works, thank you very much

3DPrintDemon, 2024-10-09:
Fantastic, nice work! Happy printing! 😎😎👍

closed 2024-10-09 by 3DPrintDemon