3DPrintDemon/Demon_Klipper_Essentials_Unified #31: Home Control - Voron Tap

weismantelf, 2024-10-09:
Hello,
I just want to know if I have to comment out the [homing_override] section
when I a using a Voron Tap?

3DPrintDemon, 2024-10-09:
Hi there, the Voron Tap documentation says to use their system with z endstop as `endstop_pin: probe:z_virtual_endstop`.

It does say you can use `safe_home` or `homing_override` but requires you to set it to home the Z axis in the middle of the bed. My `homing_override` will do this for automatically when your Z endstop is set to `endstop_pin: probe:z_virtual_endstop`.

Please see Voron Tap docs....
https://github.com/VoronDesign/Voron-Tap/blob/main/config/tap_klipper_instructions.md

NOTE: I have not tested Voron Tap with these macros as I don't have a machine with it. So please do be careful on first use. It should be fine as I can't see anything that might ruin your day here, but you never know. Maybe use an old bed or  something else to test the homing function on. You could use the `Z_ASCENDER` macro to raise the Z Axis before homing & place a bed stunt-double under the nozzle just to test. USE CAREFULLY as you will be able to top-out your machine with `Z_ASCENDER`!!!

weismantelf, 2024-10-17:
When I am using your macros, I have big problems with leveling the gantry.
I need sometimes 5 round,  so 25 retries until the gantry is leveled. 
When I am turning of your macros, I need between 2 and 4 retries.
I observed that the values of the CNC Tap are not corrected when
your macros are turned on. So I get for some probe the vale 0.81, 0.61, 031, 030, 0.28.
When the macro is turned off, the values on the same probing point are between to 0.309 and 0.315.

Have you any idea what can cause the problem?

3DPrintDemon, 2024-10-17:
I’m not sure why that would be, there’s absolutely nothing in them anywhere that would modify or alter any QGL values. 
all the system does is some setup like homing & heating to run the QGL. It uses the stock Klipper commands with nothing changed or edited that would effect the results of any probing or meshing. 

The only thing that effects offset are the filament based adjustments but those can be turned off & are not anything to do with levelling or meshing. They’re gcode offset adjustments. 

I unfortunately cannot test any operation with a Tap system as I don’t have one so I can’t tell you why it would do that or not. 

Maybe having heaters on while probing makes a difference, don’t know why it would. Maybe there’s a change in homing speed that affects the results. 

Did you add the required Tap activation macro to the homing process when using my macros as per the link I posted above? As my macros do not have that.

weismantelf, 2024-10-22:
I found the main cause of the issue, it was not your macros, it was a loose screw on the Z3 stepper.

closed 2024-10-22 by weismantelf