3DPrintDemon/Demon_Klipper_Essentials_Unified #32: Feature Req

rssw, 2024-10-09:
Hi!  Thanks a million for this great macro pack!

I've written a few macro's in the past to get things working the way I like, so I have great appreciation for what went into this.  Great job!

However, I'm surprized to find no retraction settings in the pack, when PA and Temp are already set based on each filament.  I'm sure you must be handeling this somehow.  Do you just leave those settings to Orca, or run custom startup Gcode for each filament (SET_RETRACTION ...)?  If so, why the divergent approach?

Again, great job and I've placed my settings in Orca for now, but I look forward to future developments!

3DPrintDemon, 2024-10-09:
Hi you’re welcome. I’m so glad you like my work, it’s been a long road to get this far, but worth it I feel. The recent 2.9 update has been well received. 

I personally like to leave retractions in the slicer as they can vary even in the same filament type.
Between brands, between colours, & even sometimes roll to roll. The same goes for printing temperatures. 

While it would be relatively easy to add a retractions variable to the filament type I feel the value might be a hassle or a hinderance even if you wanted to change it from roll to roll in the same filament type. 
a change from type to type should be ok but roll to roll in the same type would be problematic unless you only ever needed one value for that type. 

If it’s in the slicer you can store the value with the filament profile & it’ll bring that in directly with the filament Gcode section as you say, so eliminating that single value per type problem.

rssw, 2024-10-10:
Good to know.  I'll bow to your experience and follow the same approach.  I've got many years with an Ender under my belt, and have never had the kind of repeatability to let these differences show, but now with my Voron I'm happy to learn a few new tricks.

I happen to have installed and tested v2.8 about 2 days before your v2.9 release, but the upgrade was pretty smooth.  Very happy to have the added intelligence.  Definitely an upgrade to the base macros.

closed 2024-10-10 by rssw

3DPrintDemon, 2024-10-12:
I think it'd just build in problems is there were preset retraction values here. I'm glad you're enjoying the macros!