3DPrintDemon/Demon_Klipper_Essentials_Unified #34: Feature Request

mmrrxx00, 2024-10-22:
Would it be possible to add the ability to use the purge logos from KAMP.  Was just a thought you have a cool logo and I have seen voron and zerog logos also.

3DPrintDemon, 2024-10-24:
Thats a cool idea, I'll have to have a look at getting the demon converted to a purge line! Hopefully it's possible.

mmrrxx00, 2024-10-24:
[demon.zip](https://github.com/user-attachments/files/17508433/demon.zip)
Here is a quick step file I was able to make. It would need scaled down a lot but this way you could set the custom size for the gcode. I used the pic from your profile to create this.

stephbellemare, 2024-10-24:
@3DPrintDemon since you’re using kamp_light already, why not use the kamp purge voron logo ?

3DPrintDemon, 2024-11-01:
> @3DPrintDemon since you’re using kamp_light already, why not use the kamp purge voron logo ?

Because not everyone owns a Voron & might not want to have three little stripes to take off the bed instead of one single line. I have a Voron & I don’t want to have to remove 3 lines instead of 1 I must say. 

However if you wanted to use them it’s an easy enough change

closed 2024-11-21 by 3DPrintDemon