3DPrintDemon/Demon_Klipper_Essentials_Unified #39: Bed mesh generator wrong temperature

AndreasPlate, 2024-10-31:
Hi,

So I am going to preface this that I am not excluding an error on my side, but I am lost and resources on your amazing pack is still limited in the community.

I have a stock Sovol SV08 and, after a full system update, followed instructions to install everything. I am using Orcaslicer 2.2.0 and everything seems in order.

I am now starting to calibrate everything so decided to run the full mesh building macro and this is where I run into the issue. The bed nicely pre soaks up to printing temp then it drops back to the stock set 65c before doing the mesh. I am trying to navigate the config files to trace back why and where it does that but I am not finding it. If you have any pointers I'd be happy to try them.

3DPrintDemon, 2024-10-31:
Hi it sounds like you still have the Sovol macros included in your system & they're causing a conflict.

This is due to Sovol renaming the BED_MESH_CALIBRATE command & using it as their own macro. Which does indeed reset the bed to 65ºc before building a bed mesh.

Please confirm you have commented out line 4 in the Sv08 printer.cfg, it should look exactly like this....
`#  [include Macro.cfg]`

If it doesn't have the `#` at the start please add one then save & restart

AndreasPlate, 2024-11-01:
Thank you, that solved my issue! I'm going through the documentation but other then mentioning specific edits for macro.cfg maybe include a small section to comment it out in printer.cfg all together?

closed 2024-11-01 by AndreasPlate

3DPrintDemon, 2024-11-01:
Hi great that solved your issue. The line to comment out all Sovol macros is near the top for the “edit your printer.cfg” of the sv08 section of the documentation. 

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/Sovol_SV08_Setup/SV08_EXTRA_INSTRUCTIONS.md#edit-your-printercfg

[image]