3DPrintDemon/Demon_Klipper_Essentials_Unified #40: Bed fans error

bmberetta, 2024-11-01:
I have bed fans installed on my V2.4, and I have set up my printer.cfg correctly (I think...). When I try the _SET_BED_FAN bed test in file demon_bed_fans_monitor_v1.3.cfg, I keep getting the following error. 

Error evaluating 'gcode_macro _BED_FAN_SET:gcode': gcode.CommandError: This error is caused by no correctly named chamber_sensor or temperature controlled chamber_fan being available to the system! Check the system is setup correctly!

My relevant printer.cfg settings

```
[temperature_sensor Chamber_Temp]
sensor_type: Generic 3950
sensor_pin: PB0

[fan_generic Bed_Fans]
pin: PA4
max_power: 1.0
shutdown_speed: 0
kick_start_time: 0.5
off_below: 0.19

[output_pin DISABLE_BED_FANS]
pin: PE7 # Dummy pin
```

What am I doing incorrectly?

3DPrintDemon, 2024-11-02:
Hi! Your setup looks correct there.

Are you running a print or just sending  `_BED_FAN_SET` as a standalone command in the console?
Just to confirm, you have all the latest files installed yes? (v2.9.1)

If its a test command you'll need to send this command first...
```
_CHAMBER_SENSOR_DEFINE
```

Sorry the info at the top of the bed fans file is a bit out of date.
I'll put that on the list to update!

bmberetta, 2024-11-02:
Thanks for the quick response. I'm not running a print, just running the test to see that all works before putting the electronics bay panel back on again.

I tried the command given, but now get this error when issuing the ```_BED_FAN_SET``` command.

```
Error evaluating 'gcode_macro _BED_FAN_SET:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
```
Do I need a temperature fan setup as well?

3DPrintDemon, 2024-11-02:
Ok found an issue in the file! it is now corrected & pushed to the repo.

Please re-download the `demon_bed_fans_monitor_v1.3.cfg` file & test.

It's crazy late here now so hopefully it'll work for you. I'll pick your reply up in the morning.

bmberetta, 2024-11-02:
Working perfectly now! Thanks very much for the very quick response and resolution!

3DPrintDemon, 2024-11-02:
Hey that’s fantastic news! So glad you got it working! I love the floating bed fans macro on my 2.4, it really helps with circulation & decreased chamber heating times!

bmberetta, 2024-11-03:
Yes, the bed fans macro decreased my chamber heating times by at least 40%. Thanks for these great utilties.

3DPrintDemon, 2024-11-03:
Ohh thats great news! I'm so pleased it worked well for you, & thank you for your support! It's very much appreciated!

closed 2024-11-03 by bmberetta