3DPrintDemon/Demon_Klipper_Essentials_Unified #41: Nevermore StealthMax

mmrrxx00, 2024-11-02:
I am not sure if I am configuring something wrong or if it isn't supported yet. I am trying to setup the smart nevermore stealthmax to work with your macros. Is this a supported configuration? If not could it get added as a feature request?

3DPrintDemon, 2024-11-02:
Hi. That system has not been tested with the macros as I don’t have one sadly. So I’m afraid I can’t comment on its compatibility at the moment. If you have a link to the correct GitHub repo I can take a look for sure though. 👍

mmrrxx00, 2024-11-02:
I am not sure but it might just need to ask if someone has a smart system then adding their start commands. I am sure it is a lot more work then that but kind of what it looks like to me.  I am not that good with writing code though. 

Here is the github  for the the controller. 
https://github.com/SanaaHamel/nevermore-controller

Here is the main github for the stealthmax 
https://github.com/nevermore3d/StealthMax/tree/main

3DPrintDemon, 2024-12-03:
Hi sorry been busy with other update stuff recently. I have the issue open here to remind me about it. 
I'm glad you're finding the expansions useful for the time being