3DPrintDemon/Demon_Klipper_Essentials_Unified #42: Error evaluating 'gcode_macro _ADAPTIVE_LEVELLING:gcode'

bmberetta, 2024-11-03:
I use a Beacon probe on my V2.4 for levelling. It worked great in v2.9. After upgrading to v2.9.1, I now get the following error message 

`Error evaluating 'gcode_macro _ADAPTIVE_LEVELLING:gcode': gcode.CommandError: This error is caused by your printer not supporting this feature! Please enable Manual Levelling in the demon_user_settings.cfg file.`

I'm assuming this came about due to inclusion of Eddy support. How do I go about configuring these great utilities to allow Beacon, or to bypass the probe check?
Thanks in advance for the help.

3DPrintDemon, 2024-11-03:
Hi, thats a bit strange as that macro hasn't changed between versions.
All it does is to check what movement system your printer has (coreXY or cartesian) & if it has z-tilt or QGL in the printer.cfg, then it'll issue the appropriate levelling command for your setup.
The only way you should ever see that message is if the macros can't see either of those items in your printer.cfg, did you move them out into say a beacon.cfg or something?
You do still use a QGL printer.cfg section with beacon right?

bmberetta, 2024-11-03:
Thanks for the response and pointer to the problem. I use DK (Danger Klipper) as it has much better sensor homing features. You don't have to have a `[homing_overide]` section, which Beacon does not support. DK has a feature in which you can set different accels for both x and y axes. It is set by using `limited_corexy` as the kinematics option under the `[printer]` section. I've reverted back to `corexy` as the kinematics option and all is working correctly again. Yes, you do use QGL section in printer.cfg with Beacon.

3DPrintDemon, 2024-11-04:
Ahhh ok I see good to know that was the issue. That branch hasn’t been tested with these macros I’m afraid & they’re not setup to deal with that feature. 

Great I was gonna say, probably not the QGL section as that’s always required as far as I know. 

This system is also not tested on Beacon as its prohibitively expensive to buy one outside of the USA.

closed 2024-11-04 by 3DPrintDemon