3DPrintDemon/Demon_Klipper_Essentials_Unified #43: Sv08 Calibration routine

jeremyforan, 2024-11-03:
Hello,
Its not clear to me the steps I should be taking to dail-in my printer (SV08). I am getting failures which I believe to be related to Z-Offset.

Can you run through the list of MACROs to run after completing the installation of Demon Klipper Essentials.

- QUAD_GANTRY_LEVEL
- ??? Z_PROBE_ACCURACY_HOT
- .......

Thanks for the guidance.

3DPrintDemon, 2024-11-04:
Hi, I’m happy to help but please note this section is meant for issues with functionality of the macros, bugs etc & not really general printer setup questions. The macros use all the standard Klipper commands & processes for setting up the machine but with everything you need added to a one click macro so it effectively automates each action by setting the printer up ready for your chosen command. 

The stock SV08 probe can be of varying quality & normally is effected by thermal drift when heating up or cooling down, so this means result from print to print can vary a lot, even more so between machine to machine! Some users get better probes than others. So please keep this in mind!

You need to….

Run a full PID tune if you have a new printer.cfg. 

Make sure your z offset is correct with Z_PROBE_CALIBRATE_COLD /HOT, your choice. 

The macros will do any QGL etc for you whenever it’s needed. 

Heat soaking the inductive probe helps before you build any meshes or start a print, again the macros do that for you too if you use the auto bed mesh builder or if you have the heat soak timers installed correctly & the option set to heat soak when you start a print. 

Probe_accuracy is only used to see how your probe is working. 

Basically all you gotta do is set your z offset & tune it & the macros do the rest for you with your choice of heat soak.

3DPrintDemon, 2024-11-04:
There is nothing that changes flow rate in the macros that is not set to 100% by default. There is also nothing that directly changes the flow rate during a print. The macros shouldn’t be the cause of this. 

The adaptive pa is the only thing that changes things but unless you set that way too high it shouldn’t cause this, but you can disable that easily to see. 

You can also disable the print start flow rate command in the user settings file too if you wish to see if that makes a difference. However that command is set one time only at the start of the print & can be seen as a flow rate change in mainsail.

jeremyforan, 2024-11-08:
Hey, 
Again thanks for this great work. I really appreciate you sharing this. 

I factory reset my SV08 and the problem persisted. It turns out there was a small clog in the extruder. Did a couple of cold pulls and it resolved the issue. 

I am going to switch back to Demon.

Thanks again and I apologize for raising the issue.

jeremyforan, 2024-11-08:
resolved, user error.

3DPrintDemon, 2024-11-10:
Great you got it working! Happy printing!

closed 2024-11-10 by 3DPrintDemon