3DPrintDemon/Demon_Klipper_Essentials_Unified #44: SHAPER_CALIBRATE not working

AndreasPlate, 2024-11-04:
Hello again,

I understand the structure of your installation guide better now, and I have everything up and running on a SV08 that is running mainline klipper.

When getting around to calibration, I tried the Auto Shaper Calibration, however it quits with the error that SHARE_CALIBRATE is an unknown command.

I've tried to backtrack where it is pulling the calibration from, but this is where I am getting lost. 

If there is a difference between the stock and the mainline Klipper installation for the SV08, do you have instructions on how to proceed from here?

Thanks you in advance!

3DPrintDemon, 2024-11-04:
Hi there should’nt be anything called SHARE_CALIBRATE. There’s no such thing as that in klipper or the macro files, I assume you mean SHAPER_CALIBRATE. This is built in command to Klipper. 

Be sure you have set up the system correctly link here:
https://www.klipper3d.org/Measuring_Resonances.html

AndreasPlate, 2024-11-04:
Yes I meant SHAPER_CALIBRATE, sorry for the typo. Am I correct in saying then that the mainline Klipper installation does not have this configured by default, and I need to install and configure before running this macro?

3DPrintDemon, 2024-11-04:
That’s ok. Just good to be clear incase others see it & get confused. 

There should be information on anything needed to be set up for SHAPER_CALIBRATE in the mainline SV08 repo. If not I’d recommend raising an issue on their page, as that has nothing to do with me or their repo I’m afraid. 

…but it couldn’t hurt to run that link I posted above to make sure the system is ready to run input shaping commands.

3DPrintDemon, 2024-11-04:
Yes I get a lot of people messaging & posting issues with SV08 mainline installs. Sadly it has nothing to do with me or my macros. Glad you found the solution to your problem there. 

I’ll be sure to look at the documentation layout. 👍

closed 2024-11-04 by 3DPrintDemon