3DPrintDemon/Demon_Klipper_Essentials_Unified #49: Not an issue -- compatibility with danger klipper bev2?

Kcdc3d, 2024-11-16:
Hi there, I am a fan of what you've built here, it's a system I personally wish I had the time to do as well.. I would like to take advantage of your setup, however I am concerned of possible issues since I am using some features in DangerKlipper Bleeding Edge V2 that may be incompatible, namely MPC extruder temp control, which utilizes material type tables for specific print temps/cals, limited corexy, for differing accels per axis, i think the others may be fine, but I also use motors_sync, tmc_autotune, shake&tune so before I dive in and go through the install and test process, do you know if I will run into issues on this specific klipper branch or those added features? Nice work again!

3DPrintDemon, 2024-11-21:
Hi sorry for the delay in replying. I’m glad you like the look of the macros. 
Most stuff should be ok but I think there was someone a while ago who was using Danger Klipper & had an issue with either homing or machine levelling due to the different branch having a different entry away from mainline for that feature in the printer.cfg which the macros look for. 
I forget what exactly it was but I have a feeling it was a difference in motion type (corexy) naming or a levelling section name like z-tilt or QGL. In danger klipper it was different apparently.

Kcdc3d, 2024-11-30:
you're most likely referring to limited_corexy, which allows differing accels by axis. It has thrown up red flags in a few plugins, since that specific module isn't in their data, but all they had to do was add that new portion (limited_corexy, limited_cartesian) into their current code. link: https://dangerklipper.io/Config_Changes.html?h=limited#changes . I know it was a simple fix for one plugin (TMC_SYNC_MOTORS), but I am unsure how far you'd have to go into your work to add these in. perhaps it's an additional variable. i.e (using core_xy? {x}YES, {x}NO or similar. That is if you'd need to at all, I might try it out anyway and see how it goes. Thanks for all the hard work and taking the time to respond.

3DPrintDemon, 2024-12-03:
Hi thats ok, you're welcome. 
That was it yes. I'll have a look to see what I can do with it. As this is the second time this has been mentioned.