3DPrintDemon/Demon_Klipper_Essentials_Unified #51: stepper x and y run current stays at 18/32 after homing movement. SV08

guerillaengineer, 2024-11-19:
Hey bud, loving your macros so far. have had to fix a few things, mostly due to my having a klicky probe and purge bucket installed.

DRIVE_VARS is calling my x and y stepper run current to be at 0.8 so DUMP_TMC STEPPER=X or Y is giving 18/32 which is about half of the printer.cfg defined value of 1.4. this is post-home during a print.
It will also spit out:
`Run Current: 0.70A Hold Current: 0.70A
 Run Current: 0.79A Hold Current: 0.79A`
in the console while homing. I had another form of START_MACRO which also called stepper current like this and it read 1.39 post-homing.

I'm going to fix it by changing that 0.8 to 1.4. I don't think it will do anything bad but I only know enough jinja2 to be dangerous!

3DPrintDemon, 2024-11-19:
Ahhh ok I know why it’s doing that. 0.8 is a safety “default” value incase the define driver macro is not functioning. 
mad you’re using klicky probe with the macros you have to edit the klicky macros file to include a few macro calls so it works correctly. Namely the _DRIVER_DEFINER macro as explained here in this section of the documentation. 

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#unless-youre-using-klicky-probe

you simply need to add that line to the klicky macros & the demon_homing stuff will work correctly.

guerillaengineer, 2024-11-19:
oh dude, duh. sick, I don't think that was in the readme but I could have easily, EASILY been too stoned and skipped over it. 

you're a true champion, thank you for all of your amazing work!

3DPrintDemon, 2024-11-21:
lol no worries man, I’m sure many folks have been there! 😂

glad it’s now all working for you. Happy printing! 👍

closed 2024-11-21 by 3DPrintDemon