3DPrintDemon/Demon_Klipper_Essentials_Unified #52: Bed fan macro erroring out

Motor5622, 2024-11-21:
I keep getting this error and it stops the print
Error evaluating 'gcode_macro _BED_FAN_SET:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
This a fresh install of v2.9.1 Demon_Klipper_Essentials_Unified here's my printer.cfg for the bed fan and chamber temp

[temperature_sensor Chamber_Temp]
sensor_type: Generic 3950
sensor_pin: PA0

[fan_generic Bed_Fans]
pin: PF7
max_power: 1.0
shutdown_speed: 0
kick_start_time: 0.5
off_below: 0.19

[output_pin DISABLE_BED_FANS]
pin: PC14

Even if I set start my print already to true like below I get that error

variable_start_my_print_already: True           
variable_chamber_temp_wait: False           

I know the chamber temp and bed fans work because I can control the fans on the dashboard and the chamber temp increases if I touch it.

Motor5622, 2024-11-21:
got it working went thru the demon_bed_fans_monitor_v1.3.cfg and found this

# If you use this macro as a stand alone system:
# If you have only a chamber temp sensor & not a chamber fan you'll need to "ctrl+F" "temperature_fan chamber"
# & replace all instances EXCEPT the two in the _BED_FANS_HELPER macro with "temperature_sensor Chamber_Temp"
# Also make sure "variable_chamber_fan_enable" is set False!

Replaced "temperature_fan chamber" with "temperature_sensor Chamber_Temp" and it worked even though it is not a stand alone system

3DPrintDemon, 2024-11-21:
Hi. The issue is caused by the hardware option `chamber_fan` being enabled but you only having your system set as having a ‘chamber_sensor’. The `chamber_fan` option is for an automatic temperature controlled exhaust type fan.

As such the macros will look for that setting & adapt themselves for it, but they can only do that if what they’re looking for is there. 

What you edited will obviously work but if you swap `chamber_fan` option to `False` & set  `chamber_sensor` to `True` it would’ve worked with no edits to the files. 
Although the sensor would obviously still need the correct name in your `printer.cfg`. 

However you could use the `chamber_fan` option if you set your `printer.cfg` up to use one. 

Please see this link…
https://www.klipper3d.org/Config_Reference.html?h=temperature_fan#temperature_fan

Motor5622, 2024-11-22:
I swapped out the `bed_fans_monitor` for an unmolested one and in the hardware section I do have the `chamber_fan` set to false and the `chamber_sensor` set to true. I still am getting that error. Is this correct for the `Chamber_Temp` in the `printer.cfg`?

Motor5622, 2024-11-22:
Found it. in the `bed_fans_monitor` in the description/instructions you have it called out as `Chamber_Temp` which is what I named it, but if you do a a CTRL+F and search for `Chamber_Temp` and keeps going down, in the code it is `chamber_temp` changed my `printer.cfg` to have them in "C" and "T" to lower case and it works

3DPrintDemon, 2024-11-22:
This is really strange as I have just tested the latest files here on my Voron 2.4 350.

I normally have mine setup as using a `chamber_fan` but I modified the system to test this. I set it to use the `chamber_sensor` hardware option with the sensor called `[temperature_sensor Chamber_Temp]` as it is described in the documentation.

The system picked this up immediately & ran the _BED_FAN_SET macro without a single issue or error. It successfully controlled the bed fans' speed during the PRINT_START macro.

So I'm not exactly sure to why yours is not working correctly if your system is set up like the documentation & what you have posted above. Going back to your original comment, the error "has no attribute" means the thing the macro is looking for is not there where it's supposed to be. The macro in this instance has been told to use `temperature_fan chamber` by a helper macro that looks in your printer.cfg file to see what type of sensor you have defined. This then posts the findings for the rest of the macros to use. 

So basically somehow your system has been told it has a `temperature_fan chamber` but when the macros look for temperature information from it it's not there as you have a `temperature_sensor Chamber_Temp`. Obviously it shouldn't do this.

You can get the macros to tell you what type of sensor its setting by adding this to line 72 of the `PRINT_START` macro, at the bottom of the `INITIAL SETUP` section.

```
RESPOND TYPE=COMMAND MSG="chamber_thermal_sensor {core_vars.chamber_thermal_sensor}"
```

It will return an assigned value in the console for use with `_BED_FAN_SET`.
- 0 = default value 
- 1 = "temperature_fan chamber" found
- 2 = "temperature_sensor Chamber_Temp" found
- 3 = neither "temperature_fan chamber" or "temperature_sensor Chamber_Temp" were found

OR.....
to simply test the system without starting a print you can call this macro in the console.
``` 
CHAMBER_SENSOR_DEFINE
```
It won't show or appear to do anything that you can see, this is normal.

Then add this macro to the my_macros.cfg (or other) file & call it after running the above to see what sensor the system thinks you have...
```
[gcode_macro SENSOR]
gcode:

    RESPOND TYPE=COMMAND MSG="chamber_thermal_sensor {core_vars.chamber_thermal_sensor}"
```

3DPrintDemon, 2024-11-22:
This is really strange as I have just tested the latest files here on my Voron 2.4 350.

I normally have mine setup as using a `chamber_fan` but I modified the system to test this. I set it to use the `chamber_sensor` hardware option with the sensor called `[temperature_sensor Chamber_Temp]` as it is described in the documentation.

The system picked this up immediately & ran the _BED_FAN_SET macro without a single issue or error. It successfully controlled the bed fans' speed during the PRINT_START macro.

So I'm not exactly sure to why yours is not working correctly if your system is set up like the documentation & what you have posted above. Going back to your original comment, the error "has no attribute" means the thing the macro is looking for is not there where it's supposed to be. The macro in this instance has been told to use `temperature_fan chamber` by a helper macro that looks in your printer.cfg file to see what type of sensor you have defined. This then posts the findings for the rest of the macros to use. 

So basically somehow your system has been told it has a `temperature_fan chamber` but when the macros look for temperature information from it it's not there as you have a `temperature_sensor Chamber_Temp`. Obviously it shouldn't do this.

Add this macro to the my_macros.cfg (or other) file & restart 
```
[gcode_macro SENSOR]
gcode:

    RESPOND TYPE=COMMAND MSG="chamber_thermal_sensor {core_vars.chamber_thermal_sensor}"
``` 
You can get the macros to tell you what type of sensor its setting by adding this to line 72 of the `PRINT_START` macro, at the bottom of the `INITIAL SETUP` section.

```
SENSOR
```

It will return an assigned value in the console for use with `_BED_FAN_SET`.
- 0 = default value 
- 1 = "temperature_fan chamber" found
- 2 = "temperature_sensor Chamber_Temp" found
- 3 = neither "temperature_fan chamber" or "temperature_sensor Chamber_Temp" were found

OR.....
to simply test the system without starting a print you can call this macro in the console.
``` 
CHAMBER_SENSOR_DEFINE
```
It won't show or appear to do anything that you can see, this is normal.

Then call it after running `CHAMBER_SENSOR_DEFINE`  to see what sensor the system thinks you have...
```
SENSOR

```

Motor5622, 2024-11-23:
So I tried both ways to get the assigned value of the sensor and both ways I get the same result `Unknown command:"_SENSOR"` also when I turned the printer on tonight I am back to getting the original error I have not changed my printer.cfg it still is

[temperature_sensor Chamber_Temp]
sensor_type: Generic 3950
sensor_pin: PA0

[fan_generic Bed_Fans]
pin: PF7
max_power: 1.0
shutdown_speed: 0
kick_start_time: 0.5
off_below: 0.19

[output_pin DISABLE_BED_FANS]
pin: PC14

I have floating bed fan set to true
`variable_start_my_print_already: False`          
`variable_chamber_temp_wait: True`   

`variable_post_print_cool: 25` 

So I'm not sure what is missing I've been thru the instructions so many times I almost have it memorized

3DPrintDemon, 2024-11-23:
The unknown command on the `_SENSOR`macro, I’m assuming you pasted it into an included & active .cfg file for use with your system as described above? 
As if not this is Klipper’s way of telling you that this macro is not in your system & it is “unknown”. 

Please double check you pasted the above code block into an included .cfg file, or it won’t work as it’s not part of the standard macro pack.

Motor5622, 2024-11-24:
I had it in the `macro_cfg` after reading your post I tried it in my `printer.cfg` and ran both testing ways and it gave me  `chamber_thermal_sensor 2` could not get it read the `macro.cfg` would up redoing and renaming  `my_macro.cfg`  to get the test to work with it not in my printer.cfg and sensor 2 is what it reports back. This is what I get when starting a print now which is more than it was it use to just stop at the error

8:15 PM Bed Fans Monitor: Stopped, fans off
8:15 PM Restoring parts fan speed
8:15 PM Resuming print!
8:15 PM Bed Fans Monitor Active
8:15 PM Printer not homed
8:15 PM Error evaluating 'gcode_macro _BED_FAN_SET:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
8:15 PM Error evaluating 'gcode_macro _BED_FAN_SET:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'temperature_fan chamber'
8:14 PM ABS File Detected Hotend Preheat: 190c
8:14 PM chamber_thermal_sensor 2
8:14 PM pressure_advance: 0.035000
              pressure_advance_smooth_time: 0.030000
8:14 PM PRINTER CHECK: PASSED
8:14 PM File selected

Motor5622, 2024-11-25:
When I comment out lines 587 and 588 it still did not work, but something I found is the `bed_fans_monitor` I had downloaded on 11/18 lines 587, 588 were the same as lines 590, 591 so I got the one from 11/25 and I worked with no edits to any file. The file from 11/18 was a complete zip file and the one from 11/25 was just `bed_fans_monitor` on the main page. Don't know why they were different if it was something I changed and didn't know or what, all I do know is it now works. Thanks for the help and some nice macros.

3DPrintDemon, 2024-11-26:
Ah ok maybe an error correction there then in that case. Either way glad it's working now, you're most welcome. Happy printing.

closed 2024-11-26 by 3DPrintDemon