3DPrintDemon/Demon_Klipper_Essentials_Unified #53: Unable to home

royalfifty8, 2024-11-22:
Getting this message after trying to home all axes  ~~~  Unknown command: "SET_KINEMATIC_POSITION"   then followed this message 3 times in red ~~~ "Must home axis first: 0.000 0.000 10.000 [0.000]"

3DPrintDemon, 2024-11-22:
Hi please be sure to define the `FORCE_MOVE` section in your printer.cfg as per the documentation.
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#define-force-move-section

royalfifty8, 2024-11-22:
Ah, that's it. Don't know how I missed it. Thank you kindly! Have been using Demon macros for months now. Love em. Cheers mate!

closed 2024-11-22 by royalfifty8

3DPrintDemon, 2024-11-22:
That’s ok. It’s an easy one to miss for sure! 
So glad you like my work & enjoy the macros. 
you’re welcome. 😉