3DPrintDemon/Demon_Klipper_Essentials_Unified #54: carthographer touch

craeke, 2024-11-23:
hi 

first off its very good and i love it.

the only problem i have is when using touch for carthographer it does not do the touch before print? 
i have my scanner set to touch in my printer.cfg am i missing something? 
i assume it will be the same as normal klipper? that it wil calibrate the z offset? 

i want to change plate or nozlle and want it to calibrate the z offset like it does with normal klipper.

is that possible? 

greetings

3DPrintDemon, 2024-11-24:
Hi, if you have the `scanner` section in your `printer.cfg` & you also have `calibration_method` set as `touch` it should work. However I have added a M400 command to the file now, so the system will wait to do the "touch" until all previous commands have been completed. 

Please re-download this file & try again.
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_print_start_end_v2.9.1.cfg

craeke, 2024-11-24:
wil try it i notice when i coompare the files the m400 command in the mesh section was missing in mine.
maybe thats why? i need to wait for my print to finish to try.

any idea why i get The value 'BedFans' is not valid for FAN?

3DPrintDemon, 2024-11-24:
Hopefully it'll work, sorry I can't test it as I don't have a cartographer.

Do you have bed fans enabled in your hardware options in the demon_user_settings file? It looks like you've called your parts cooling fan BedFans.

craeke, 2024-11-25:
its not working, i need to at CARTOGRAPHER_TOUCH at the end off mesh section for it to work

3DPrintDemon, 2024-11-25:
There are CARTOGRAPHER_TOUCH commands after meshing options that are dependent on the sections & options you have in your printer.cfg, like I mentioned before. Sadly I don’t have a cartographer so I can’t test it sadly. 

You could try editing line 256 of the demon_print_start_end.cfg to this if you’re using adaptive meshing…
```
{% if printer.configfile.settings.calibration_method == 'touch'  %}
```
Plus line 310 if you’re not.

3DPrintDemon, 2024-11-25:
Ah ok this should be working but not obviously, probably due to a path error.

Try 
```
{% if printer.configfile.settings.scanner.calibration_method == 'touch'  %}
```

all I can do is suggest things to try as I can’t test this with a probe I don’t have I’m afraid.

craeke, 2024-11-25:
yeah its alright im willing to test it no problem

closed 2024-12-01 by 3DPrintDemon