3DPrintDemon/Demon_Klipper_Essentials_Unified #55: Scrubber Clean Pattern

patofoto, 2024-11-24:
I am moving to a non-metal-brush option to clean my nozzle because I am integrating a Beacon probe into my Stealthburner. I have landed on this model:

[https://github.com/scheffield/nozzle-cleaner/tree/main?tab=readme-ov-file]

Currently, your macro moves the head in a random sequence on moves given the variables set by the user. With this scrubber I just need moves on the X axis and not on Y. I have set my variables this way:

```
variable_clean_min_x: 102                        # Min safe X axis for random clean movements - SV08 Demon Purge Bucket Xmin 303
variable_clean_max_x: 112                        # Max safe X axis for random clean movements - SV08 Demon Purge Bucket Xmax 335
variable_clean_min_y: 348                        # Min safe Y axis for random clean movements - SV08 Demon Purge Bucket Ymin 356
variable_clean_max_y: 349                        # Max safe Y axis for random clean movements - SV08 Demon Purge Bucket Ymax 362

variable_start_x: 102                            # Start cleaning postion for X - SV08 Demon Purge Bucket X 294, or for stock setup use 316 
variable_start_y: 349                            # Start cleaning postion for Y - SV08 Demon Purge Bucket Y 359, or for stock setup use 359
variable_start_z: 7.6                            # Start cleaning postion for Z - SV08 Demon Purge Bucket Z 3, or for stock setup use 0.5

variable_pass_count: 24                          # Number of cleaning pass actions
variable_pass_spd: 50                            # Pass speed

variable_end_position_x: 112                     # End cleaning position for X - SV08 Demon Purge Bucket X 320, or for stock setup use 351
variable_end_position_y: 349                     # End cleaning position for Y - SV08 Demon Purge Bucket Y 359, or for stock setup use 359

variable_park_x: 175                             # - SV08 Demon Purge Bucket X 344
variable_park_y: 345                             # - SV08 Demon Purge Bucket Y 345
variable_park_z: 25                              # - SV08 Demon Purge Bucket Z 25  
```

If I set `variable_clean_min_y` and `variable_clean_max_y` to the same number to only move in X I get an error. I have set mine for now with a 1mm distance from each other. Is there a better way of using this scrubber?. Am I missing something?. Here is the macro that this scrubber uses just in case it helps you:

[clean_nozzle.cfg.zip](https://github.com/user-attachments/files/17893336/clean_nozzle.cfg.zip)

I love your macros!!!

3DPrintDemon, 2024-11-24:
Hi there, so glad you like my work! This has been something I've been thinking about on & off, so I've posted a mini update to add this feature for you!

I've added new features where you now have 2 different cleaning modes, `Linear cleaning Mode` & `Random_Cleaning_Mode`. You can mix & match them & choose to use either on one or both Axes! 

For example you can have random movements on the X axis but on a single position on the Y axis. Or you can have simple full linear cleaning & not use any random movements. It also has a choice if your cleaner is along the Y axis instead of the X axis.

To use please download:

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_clean_load_v1.8.cfg
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_core_assets_v1.3.3.cfg
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_user_settings_cleaner_variables_v1.1.cfg

patofoto, 2024-11-25:
OMG!!! So cool. Super kind of you. Will this be incorporated in the next release or should I keep these in a safe place?. You rock!!!

3DPrintDemon, 2024-11-25:
You’re most welcome! Let me know how you get on with the new feature. It’s seemed to work well on my Voron here. 
the files are already part of the main release. The component files have been updated to include this feature.

3DPrintDemon, 2024-11-26:
Hi, actually I don’t think anything came through. That’s weird! Best send it through on messenger in that case.

patofoto, 2024-11-26:
Sorry, messenger?. Message you to your GitHub user?. Don't seem to find out how.

closed 2024-12-01 by 3DPrintDemon