3DPrintDemon/Demon_Klipper_Essentials_Unified #56: _CONDITIONAL_CLEAN add exceptions for cartographer touch

OoSTARTERoO, 2024-11-26:
I love your macros! this is gold!

However, I would like to make a small request on "_CONDITIONAL_CLEAN " macro.
For the cartographer touch to work properly, the nozzle should be cleaned beforehand.

However with current implementation of "_CONDITIONAL_CLEAN", Since cartographer is a virtual endstop, It won't execute the clean macro

`[gcode_macro _CONDITIONAL_CLEAN]
gcode:
  {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}
  
  {% if start_vars.nozzle_cleaner == True %}
    {% if printer.configfile.settings.stepper_z.endstop_pin != 'probe:z_virtual_endstop' %}
      CLEAN_NOZZLE
    {% endif %}
  {% endif %}`
  
  therefore, I think it should include exceptions for cartographer touch / beacon touch

OoSTARTERoO, 2024-11-26:
maybe modify this line
`{% if printer.configfile.settings.stepper_z.endstop_pin != 'probe:z_virtual_endstop' %}`

to include additional case, when scanner calibration method is "touch" would work
```
{% if printer.configfile.settings.stepper_z.endstop_pin != 'probe:z_virtual_endstop' or
          ('scanner' in printer and printer.configfile.settings.scanner.calibration_method == 'touch') %}
```

3DPrintDemon, 2024-11-26:
good point, I can add that for sure. Would've been my exact solution there too for this.