3DPrintDemon/Demon_Klipper_Essentials_Unified #57: Load filament and unload filament default temperature

javilopezalarcon, 2024-11-26:
Good evening, first of all, congratulate you on the work you have here, in a wonderful way. 

I wondered if there would be any possibility of in the macro load filament and unload filament, if the value of the temperature that by default takes 250 degrees, could be a variable that could be edited, or a number like 200 degrees for example or a material and that takes the value of the, pla, petg.. I'll explain the problem to see what solution could be the best. In KlipperScreen in the extrude menu there is load and unload, but there is no fringe to choose temperature degrees or anything, if the default was editable with a variable, I could choose the one I use the most for example and that would work just by clicking on it and use the default that would take the value of a variable.
As you can see, could it be done or do you see some other better way? Best regards

javilopezalarcon, 2024-11-27:
Hi there. Thanks a lot for the reply. I'm really enjoying your macros, and learning a lot too :-)

I understand that KlipperScreen is out of your scope. The truth is that it would be ideal if they fixed it in that part, but we don't know when it will arrive :-(

What I propose, as long as it seems good to you, is that this default temperature of 250º be parameterizable. For example a new variable called:

#FILAMENT CHANGE/LOAD/UNLOAD MACRO SETTINGS
...
variable_clean_load_default_temp: 250

And then use this line:
{% set temp = params.TEMP|default(start_vars.clean_load_default_temp)|int %}

In all the macros in the file:
demon_clean_load_v1.8.cfg

For example, I usually print PLA in 99% of the cases 250º is too much.

Thank you very much for everything, and have a nice day

3DPrintDemon, 2024-11-28:
yes it would be really nice if they improved that part of their system. One day maybe.

Ahhh ok I understand now. While that does work having a variable there in the default field it actually causes the macro to be displayed with that variable in the temperature field, which is confusing & counter intuitive as you have no idea what that value is unless you go to find the variable in the user settings file, which you can't fully read in the macro field. See image.

So I think while thats a very good idea to be able to have variables set there for these the way mainsail works makes them a less desirable solution in this instance. I would recommend that if 250 is too much for your normal load/unloading requirements that you should edit the file's `|default(250)` values to say `|default(200)` for example in each of the load/unload macros. 

This way you'll have your lower default temperature but still be able to read it in the macro field & also be able to have to original functionality.
[image]

3DPrintDemon, 2024-11-28:
You're most welcome. Happy printing!

closed 2024-11-28 by 3DPrintDemon