3DPrintDemon/Demon_Klipper_Essentials_Unified #58: User hooks in macros

sermayoral, 2024-11-26:
Hi friend! I have recently discovered this repository, and I love it! I think it's super powerful and it's wonderfully done. Congratulations!

I would like to propose an improvement that I find very interesting. It involves allowing the user to add custom macro code to some of the most important macros, RatOS style.

Imagine that a user, after finishing a filament unload, wanted to hear a BEEP, maybe to send an MQTT message to his home server to perform some automation.

RatOS does this by defining a series of user macros that are called from these important macros, and which are initially empty. This way, if a user wants to add some action to that macro, he would just have to redefine that user macro in his my_macros.cfg file and do whatever he wants. See https://github.com/Rat-OS/RatOS-configuration/blob/v2.1.x/macros/user-hooks.cfg and https://github.com/Rat-OS/RatOS-configuration/blob/v2.1.x/macros.cfg

¿What do you think about it? Of course I could do the coding and make a PR, but I wanted to tell you about it first and get your opinion.

Thank you very much for everything :-)

3DPrintDemon, 2024-11-26:
Hi, welcome onboard & thank you!  I'm so glad you're enjoying my work!

Now that is a very cool suggestion, I think that'll work great for extended user customisation! I'll get to work on something & try to implement it for the next update! Could be very exciting!

sermayoral, 2024-11-26:
Well thank you so much. You are very kind. If you need help with anything, just let me know. I look forward to the next update :-)

3DPrintDemon, 2024-12-01:
Update posted. Hope it's in the spirit of what you had in mind

sermayoral, 2024-12-03:
Totally... Thank you very much!!!

closed 2024-12-03 by sermayoral

sermayoral, 2024-12-03:
Totally... Thank you very much!!! You deserve a good coffee :-)

I was surprised to see the use of trigger variables. I don't really consider it necessary, since the definition of user macros comes empty by default, which means that their execution does nothing. Only if the user overwrites them will they do something, but if the user doesn't overwrite any user macro, they do nothing. In RatOS, for example, there are no trigger variables:

[image]

So what is the purpose of trigger variables? Just out of curiosity. That's cool, though, but I find trigger variables more harmful than beneficial, since you can have a user macro overwritten and it not work because you forgot to trigger its variable. In addition to all the management of if conditions that you have had to put in the code. If you want to disable a user macro, just comment it out, don't you think?

3DPrintDemon, 2024-12-03:
Thanks very much, its very much appreciated.

Indeed the triggers are not needed for the system to function, they're there so to bring them in & out of being called quickly & easily without having to run down the file & comment in & out full macros each time. Which I personally find a little inelegant, I find it much nicer to have a "switch" for it.
Plus I think it maybe better for people who are not so familiar with macro writing, as it gives confidence, a clearly explained & simple indication of what's in & what's not. It also means any of their macros do not need to be actually touched or modified to be run or not once written & saved.
It was user friendly, comfort choice in my eyes.

sermayoral, 2024-12-03:
> Thanks very much, its very much appreciated.
>
> Indeed the triggers are not needed for the system to function, they're there so to bring them in & out of being called quickly & easily without having [...]

A very appropriate and respectable opinion. Thank you very much :-)