3DPrintDemon/Demon_Klipper_Essentials_Unified #59: Error after update

patofoto, 2024-12-01:
Getting this error:

```
5:29 PM Unable to parse '' as a literal: unexpected EOF while parsing (, line 0)
5:29 PM Unable to parse '' as a literal: unexpected EOF while parsing (, line 0)
5:29 PM PRINTER CHECK: PASSED
```

I believe I have taken care of everything from instructions

3DPrintDemon, 2024-12-01:
Ah ok sorry about that simply home the printer. It'll cure itself after first time. Working on it

patofoto, 2024-12-01:
I see. Only get it after restart then. Right?.

3DPrintDemon, 2024-12-01:
Yes it's due to the system variables file being unpopulated on first boot. Documentation updated & file added if needed.
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#define-save-variables-section

patofoto, 2024-12-01:
Even after homing the file was not created for me. I had to SSH and paste your command. It is there now and I get no errors.
Thank you.

patofoto, 2024-12-01:
[image]

This is what the file looks like after homing and QGL. Normal?

patofoto, 2024-12-02:
I also tried rebooting to see if anything got saved after shutdown. File is still as pictured above.

3DPrintDemon, 2024-12-02:
Did you also add the save_variables added in the printer.cfg yes? You have to home the printer for the file to be updated

patofoto, 2024-12-02:
I did. My `printer.cfg` only has all my `[include xxxxx]` declarations and then I have a `350_voron_m8p_v2.cfg` as the file I use to save my printer settings. I have included it in my `350_voron_m8p_v2.cfg`.  Don't think that is the problem, right?.

3DPrintDemon, 2024-12-02:
If the `save_variables` section is not working, you'd get a error in the console.
```
UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'save_variables'
```

Do you get a console message on boot:
```
Z Tracker: Last known Z height was 0
```
Have you homed the printer? 
Have you disabled the steppers?

Are you running Klicky Probe?

patofoto, 2024-12-02:
-No console errors. 

-Do get:
`Z Tracker: Last known Z height was 0`

-Have homed the printer several times

-Have not disabled steppers (not sure what you mean here).

-I am using Klicky Probe (until tomorrow when I hopefully get my Beacon 3D probe).

3DPrintDemon, 2024-12-02:
ahhh ok I thought so. I've just been updating the documentation 

please see this.....

https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#unless-youre-using-klicky-probe

You'll need to add a few lines to their _Home_Z_ macro

patofoto, 2024-12-02:
Just to confirm:

```
[gcode_macro _Home_Z_]
gcode:
    {% set z_endstop_x = printer["gcode_macro _Probe_Variables"].z_endstop_x %}
    {% set z_endstop_y = printer["gcode_macro _Probe_Variables"].z_endstop_y %}
    {% set safe_z = printer["gcode_macro _User_Variables"].safe_z|float %}
    {% set travel_feedrate = printer["gcode_macro _User_Variables"].travel_speed * 60 %}    {% set z_drop_feedrate = printer["gcode_macro _User_Variables"].z_drop_speed * 60 %}
    {% set verbose = printer["gcode_macro _User_Variables"].verbose %}

    _entry_point function=Home_Z

    # if x and y are not homed yet, raise error
    {% if not 'xy' in printer.toolhead.homed_axes %}
        { action_raise_error("Must Home X and Y Axis First!") }
    {% else %}
        _KlickyDebug msg="_Home_Z_ XY Axis homed"
        {% if not 'z' in printer.toolhead.homed_axes %}
            {% if verbose %}
                { action_respond_info("Resetting Z position to zero") }
            {% endif %}
             _KlickyDebug msg="_Home_Z_ Z not homed, setting position as X=Y=Z=0"
            SET_KINEMATIC_POSITION Z=0
        {% endif %}

        # Move tool to safe homing position and home Z axis
        # location of z endstop
        _KlickyDebug msg="_Home_Z_ moving to Z endstop position G0 X{z_endstop_x} Y{z_endstop_y} F{travel_feedrate}"
        G0 X{z_endstop_x} Y{z_endstop_y} F{travel_feedrate}
        _KlickyDebug msg="_Home_Z_ Homing Z G28 Z"
        G28 Z0
        _KlickyDebug msg="_Home_Z_ toolhead too low, raising it to {safe_z}mm from {printer.gcode_move.gcode_position.z}mm"
        G0 Z{safe_z} F{z_drop_feedrate}

    {% endif %}

    _exit_point function=Home_Z

    M400
    _SAVE
    {% if start_vars.z_tracker_reduced_monitoring == False %}
     UPDATE_DELAYED_GCODE ID=_ACTIVE_Z_MONITOR DURATION=1  
    {% endif %}
```

3DPrintDemon, 2024-12-02:
yes correct but please also add

At the very top of the [gcode_macro _Home_Z_] add at line 837
```
{% set start_vars = printer["gcode_macro _START_VARIABLES"] %}
```
late update

patofoto, 2024-12-02:
Copy.

```
[gcode_macro _Home_Z_]
gcode:
    {% set start_vars = printer["gcode_macro _START_VARIABLES"] %}
    {% set z_endstop_x = printer["gcode_macro _Probe_Variables"].z_endstop_x %}
    {% set z_endstop_y = printer["gcode_macro _Probe_Variables"].z_endstop_y %}
    {% set safe_z = printer["gcode_macro _User_Variables"].safe_z|float %}
    {% set travel_feedrate = printer["gcode_macro _User_Variables"].travel_speed * 60 %}    {% set z_drop_feedrate = printer["gcode_macro _User_Variables"].z_drop_speed * 60 %}
    {% set verbose = printer["gcode_macro _User_Variables"].verbose %}

    _entry_point function=Home_Z

    # if x and y are not homed yet, raise error
    {% if not 'xy' in printer.toolhead.homed_axes %}
        { action_raise_error("Must Home X and Y Axis First!") }
    {% else %}
        _KlickyDebug msg="_Home_Z_ XY Axis homed"
        {% if not 'z' in printer.toolhead.homed_axes %}
            {% if verbose %}
                { action_respond_info("Resetting Z position to zero") }
            {% endif %}
             _KlickyDebug msg="_Home_Z_ Z not homed, setting position as X=Y=Z=0"
            SET_KINEMATIC_POSITION Z=0
        {% endif %}

        # Move tool to safe homing position and home Z axis
        # location of z endstop
        _KlickyDebug msg="_Home_Z_ moving to Z endstop position G0 X{z_endstop_x} Y{z_endstop_y} F{travel_feedrate}"
        G0 X{z_endstop_x} Y{z_endstop_y} F{travel_feedrate}
        _KlickyDebug msg="_Home_Z_ Homing Z G28 Z"
        G28 Z0
        _KlickyDebug msg="_Home_Z_ toolhead too low, raising it to {safe_z}mm from {printer.gcode_move.gcode_position.z}mm"
        G0 Z{safe_z} F{z_drop_feedrate}

    {% endif %}

    _exit_point function=Home_Z

    M400
    _SAVE
    {% if start_vars.z_tracker_reduced_monitoring == False %}
     UPDATE_DELAYED_GCODE ID=_ACTIVE_Z_MONITOR DURATION=1  
    {% endif %}
```

3DPrintDemon, 2024-12-02:
ok great that "should" do it

patofoto, 2024-12-02:
I think it works!. I now get:

```
[Variables]
last_known_z = 30.0
```

in the `demon_vars.cfg` file.

3DPrintDemon, 2024-12-02:
nice work!! Yea you got it!!
If that value is updating it woking!

patofoto, 2024-12-02:
Thank you. All good. Hopefully my Beacon Probe install won't be too much trouble. Have you heard of anyone using it with your macros?

3DPrintDemon, 2024-12-02:
You're most welcome! 
I think a few folks have been thinking about it that have contacted me, possibly more that are using it that haven't contacted me or posted. 

I know a few people with Cartographers are & I have added some support for them, BTT Eddy is also fully supported here. However I've not actually had much feedback for Beacon & I sadly don't own one myself.

patofoto, 2024-12-02:
Got it. I will report back my experiences in case you are interested. Have joined the Annex Engineering Discord channel, the makers of the Beacon. Hopefully I can get it working with no major issues. 

I believe it is very similar to the Cartographer. Fingers crossed.

3DPrintDemon, 2024-12-02:
Cool! Yes please do let me know how it goes & if you need anything adding for Beacon support! I'm sure it'll be great!

closed 2024-12-03 by 3DPrintDemon