3DPrintDemon/Demon_Klipper_Essentials_Unified #60: updating

craeke, 2024-12-02:
im sorry if im stupid, but how do i update? 
i have the 2.9.2 now.
i see a new release is it needed to update? if so how? 

thanks

3DPrintDemon, 2024-12-03:
Hi, not stupid at all. Good question!
The best way at the moment is to go into your config folder & check the box next to the main Demon directory & download it all to your pc so you have a recent backup before you change anything. Then place it in a safe directory just incase.

Then the easiest way to get all the new files at once is to delete the whole folder & run the SSH install command again to download the latest version.

Then simply transpose your user settings & clean settings over to the new file as there were some changes.

I would recommend getting VScode as you can directly compare both old & new settings files together with highlighted changes & simply copy the changes into your old files & update the version number variable in the file & file name. Then drag & drop the edited file into the new version directory, all done!

I would very much like to get a script to automate the process. It's on the to do list! 
...Until then though, it's a manual update.

sermayoral, 2024-12-05:
Hi! I can tell you how I update, and I do it through Moonraker, which also lets me know when there is a new version. All you have to do is add the following configuration to the moonraker.conf file

```
[update_manager Demon_Klipper_Essentials_Unified]
type: git_repo
path: ~/printer_data/config/Demon_Klipper_Essentials_Unified
origin: https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified.git
is_system_service: False
managed_services:
  klipper
  moonraker
```

The important thing here is not to mess up the original repository by modifying it, as happens with the mainsail.conf file. You must create a my-macros.conf file, for example, import all the Damon macros into it and overwrite the variables and macros that interest you.

See https://raw.githubusercontent.com/sermayoral/ender3pro-klipper/refs/heads/main/my-macros.cfg and https://raw.githubusercontent.com/sermayoral/ender3pro-klipper/refs/heads/main/my-variables.cfg as examples.

This way, Moonraker will notify you when there is an update, and you can update it with a click, always checking the breaking changes.

patofoto, 2024-12-09:
@sermayoral Amazing Sergio!. Super elegant solution to updates. Hopefully this could be implemented in the main branch. Will try to understand how you did things to be able to implement this myself.

3DPrintDemon, 2024-12-09:
I have had the Moonraker update manager method of on my other machines here in the past, but as from an authors point of view it’s proved a bit of a hassle as I wanted different versions & different test macros on each, but I could look a implanting it in the readme for people to try as users.

patofoto, 2024-12-10:
Good idea. A bit of a guide of sorts. 

I have cloned the repo and started my own branch with my settings. Hoping to see if I can keep up to date this way a bit easier.

closed 2024-12-23 by 3DPrintDemon