3DPrintDemon/Demon_Klipper_Essentials_Unified #61: Beacon Probe Integration

patofoto, 2024-12-02:
Just installed my Beacon Probe. It seems to be working properly with your macros. Will continue testing and report here.
I wanted to ask a question. In the user settings file there is a line for the Eddy:

`variable_auto_reset_eddy_offset: True            # When using USB Eddy SET_Z_FROM_PROBE the value should not be stored, set True to reset it to 0 on PRINT_END`

I am not using Eddy but assume that this applies to me too, right?. Set to `True`?.

3DPrintDemon, 2024-12-03:
Hi! Great to hear that it seems to be working so far!

Unfortunately at the moment that option checks if you have a USB Eddy in your system, if not it won't remove the offset. Let me know if its something you'd like for Beacon too.

patofoto, 2024-12-03:
Not sure if it's necessary. Why would I need to remove the offset?. My current offset is giving me good results after reseting firmware or rebooting printer.

3DPrintDemon, 2024-12-03:
you probably won't, the btt eddy gets the offset via a macro through temp compensation etc. Beacon won't have that

patofoto, 2024-12-05:
Closing this as I think Beacon works very well with your macros. So far have been printing for three days with absolutely no issues. Have a small request but will message you directly.

closed 2024-12-05 by patofoto