3DPrintDemon/Demon_Klipper_Essentials_Unified #63: Sovol SV08 Touch Display - Z Offset error

StefanKruk, 2024-12-03:
When using the Sovol Touch Display together with the Sovol SV08 and I want to use the "Calibrate Z Offset" I get the following error:

Unknown command:"_DELAY_CALIBRATE"

The gcode command is in the Macro.cfg
but as stated in https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/Sovol_SV08_Setup/SV08_EXTRA_INSTRUCTIONS.md I needed to coment the include out.

is there a Solution for this Problem already known?

3DPrintDemon, 2024-12-03:
Hi this is not a problem its a choice not to use the auto Z feature on the SV08 as it is unreliable at best & at worst has the possibility to damage your bed & your machine when it gets the auto readings wrong. I would strongly recommend not to use it. I would suggest using the paper method as per the default Klipper calibration.

However if you really want to use it....

Paste these original Sovols macro into your `My_Macros.cfg` file.
```
[gcode_macro _CALIBRATION_ZOFFSET]
gcode:
    M117 Calibrate Offset
    QUAD_GANTRY_LEVEL
    M140 S65
    G4 P500
    CLEAN_NOZZLE
    G4 P500
    M117 zoffset calibration
    Z_OFFSET_CALIBRATION

[delayed_gcode _auto_zoffset]
gcode:
    SAVE_VARIABLE VARIABLE=offsetadjust VALUE={'%05.2f' % (0)}
    _CALIBRATION_ZOFFSET
    M23 /.zoffset_test.gcode
    M24

[gcode_macro _Delay_Calibrate]
gcode:
    UPDATE_DELAYED_GCODE ID=_auto_zoffset DURATION=1.0
```

You will still need their save variables file in your system. 

This Should allow its use again. But I wouldn't recommend it & I have not tested the use of this with my macros. So you're do so at your own risk.

StefanKruk, 2024-12-06:
understood, thanks for clarification. I thought maybe it was reused for Demon Klipper Essentials to start their Z Offset calibration