3DPrintDemon/Demon_Klipper_Essentials_Unified #66: Filament ID issue

xanxaz, 2024-12-10:
Hi, Just configured everything but fighting the Filament type Identification, All set in ORCA ( ASA ) but Preheat it will detect PLA filament. 
I

PLA File Detected Hotend Preheat: 160c
Running Low Temp Chamber Timer via Filament Type

3DPrintDemon, 2024-12-10:
Hi, if your slicer start Gcode is not passing the information correctly to the macros then this can happen.

You could have the filament type set correctly in Orca (as you do) & the macros are there waiting for the info when you start a print but it doesn't make to them due to the slicers start gcode not being quite correctly formatted.

Please double check your `Machine start G-code` is input exactly as below. If not you can copy it directly by using the double box icon on the right of the code block below (as I'm sure your already knew) to ensure it's correct. 
```
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
M104 S0
M140 S0
PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[hot_plate_temp_initial_layer] LAYER=[layer_height] FILAMENT=[filament_type] EXCLUDE=[exclude_object] SURFACE="[curr_bed_type]" OAPA=[adaptive_pressure_advance]
```

3DPrintDemon, 2024-12-10:
Hopefully that will fix the problem.

xanxaz, 2024-12-10:
It works!! Thank you!!

3DPrintDemon, 2024-12-10:
Fantastic news, Im so glad that did the trick! Happy printing!

closed 2024-12-10 by 3DPrintDemon