3DPrintDemon/Demon_Klipper_Essentials_Unified #70: klippain-shaketune

Bata3, 2024-12-14:
Hi, when using any macro from the klippain-shaketune plugin, it turns off and prints with an error.
```
"Internal error on command:"_COMPARE_BELTS_RESPONSES"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown"
```
I know your macro contains shaper calibrate, but I need to use the klippain stuff. Is there a way to implement macros from shaketune so that it doesn't throw an error? thank you very much.

3DPrintDemon, 2024-12-14:
Hi the Demon macros will have no influence on the use of shake tune, they’re simply macros. The Demon macros that use shaper calibrate simply call the system feature so you can have many different macros using the same system feature without issue, it’s not exclusive as long as they’re not named exactly the same thing.

Bata3, 2024-12-14:
Ok thanks for the information, I searched a bit and the problem is in the new version of the klipper, I guess they will fix it soon.

3DPrintDemon, 2024-12-14:
Ah ok yes they recently changed some things to do with the resonance testing. Hopefully Shake Tune can get their system updates soon.

closed 2024-12-14 by 3DPrintDemon