3DPrintDemon/Demon_Klipper_Essentials_Unified #71: CARTOGRAPHER_TOUCH Not working

mohitngpr, 2024-12-14:
Hi,

I'm a big fan of work! I was using version 2.9.3 and really enjoyed it until yesterday. However, after updating the Cartographer Touch firmware to 5.1.0, I encountered an issue. The print start function no longer recognizes the CARTOGRAPHER_TOUCH command.

To troubleshoot, I rolled back to Essential 2.8, and everything is working fine again. I would greatly appreciate it if you could look into this issue and let me know if there are any fixes or suggestions.

3DPrintDemon, 2024-12-14:
Hi there. If they updated the their system there could be an issue with what they changed. I’ll have a look to see if I can see what it was. Version 2.8 doesn’t have any Cartographer Touch commands in the macros. So would seem a little strange.

3DPrintDemon, 2024-12-14:
Found the issue. They removed the calibration method entry in their setup. 

To fix this go into the demon_print_start_end file & edit lines 293 & 347 from:
```
    {% if 'scanner' in printer and printer.configfile.settings.scanner.calibration_method == 'touch' %}
```

To read now read:
```
    {% if 'scanner' printer %}
```

This should allow use of v2.9 onward

mohitngpr, 2024-12-15:
In Version 2.8, I had manually added the CARTOGRAPHER_TOUCH command to the print start. Version 2.9 is working flawlessly after the update. Thanks a lot!

3DPrintDemon, 2024-12-17:
You're welcome.

Ah ok, that would explain it! Glad its working now!

closed 2024-12-17 by 3DPrintDemon