3DPrintDemon/Demon_Klipper_Essentials_Unified #72: Timelapse

subparstar, 2024-12-15:
Have you tried moonraker timelapse? I keep getting all kinds of wonky things happening like pauses and weird flow issues

3DPrintDemon, 2024-12-15:
Hi I’m not sure how or why the macros would cause any problems for Timelapse. The macros don’t set any automated pauses or flow changes during the middle of the print. They shouldn’t have any interaction with Timelapse. 

Have you tried turning off Timelapse to see if you still get the weirdness? 

you could also turning off any features you have enabled in the macros, like bed fans, or chamber heaters, auto stop fans on pause, or try reducing the Z tracker monitoring. 
lastly you try disabling the macros entirely to see if the problem goes away or not.

3DPrintDemon, 2024-12-15:
I have just downloaded & installed the latest version of the macros onto a test machine, along with Timelapse. 
They both worked perfectly happily together for 3 separate prints over a couple of restarts with no unexpected pauses or random flow changes, & without any other events that might indicate something was not working well. The prints & Timelapse's were ok.

This would lead me to believe this is possibly a machine specific problem, I'd recommend following the above fault finding suggestions & hopefully you'll be able to locate the issue. Let me know what you find.

subparstar, 2024-12-15:
Ok, so I wiped my raspberry pi and started fresh install on everything. I haven't gone through the settings yet but I did a "machine level cold" and I got an error "extruder not hot enough" error and the hotend parked in the back right corner. Any idea what would cause that?

3DPrintDemon, 2024-12-15:
Ah ok, that should hopefully cure that issue then. As for the "extruder not hot enough" error, could be anything at this point tbh as thats not much info to go on. Was there any other messages in console? Maybe there's a step in the setup not yet completed that is causing this. I know that macro works as I've just pressed it here & it completed ok.

subparstar, 2024-12-15:
It seems like it was because "My_Sensor" in the "Miscellaneous" tab in Mainsail was activated. I turned it off and it seems to be working better now. I can't find any instance of this in the settings or config

3DPrintDemon, 2024-12-17:
Ok great you found the issue! Thanks for letting me know. Closing the issue now. Happy printing!

closed 2024-12-17 by 3DPrintDemon