3DPrintDemon/Demon_Klipper_Essentials_Unified #73: Bed fan config error

royalfifty8, 2024-12-15:
I keep getting this issue. I dont use this macro becuase i have no bed fans. Everything was working before now it not. 

File contains no section headers.
file: '/home/biqu/printer_data/config/./Demon_Klipper_Essentials_Unified/demon_bed_fans_monitor_v1.3.cfg', line: 219
'variable_readback_value: 0\n'

3DPrintDemon, 2024-12-15:
Thats a weird one. Please download the latest version of this file & see if it persists.
https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/demon_bed_fans_monitor_v1.3.cfg

It appears the system sees that variable as unattached to a macro, it can only be that way if the macro section header has been deleted, commented out or modified in some way.

royalfifty8, 2024-12-15:
That worked. Much appreciated for the timely response. As always, I cant thank you enough! Cheers.

closed 2024-12-15 by royalfifty8

3DPrintDemon, 2024-12-15:
Awesome! Glad that helped, you're most welcome. Happy printing.