3DPrintDemon/Demon_Klipper_Essentials_Unified #74: Option 'gcode' in section 'gcode_macro _START_VARIABLES' must be specified

Graf911, 2024-12-17:
I'm new to printer setup. I did everything according to the instructions, but the printer does not work. I'm asking for help. What am I doing wrong?
[demon_user_settings_v2.9.3.TXT](https://github.com/user-attachments/files/18168698/demon_user_settings_v2.9.3.TXT)

Graf911, 2024-12-17:
[image]

Graf911, 2024-12-17:
the file name should be "demon_user_settings.cfg" or "demon_user_settings_v2.9.3.cfg" ?

3DPrintDemon, 2024-12-17:
Yes the image shows it there but the error message is showing that somehow it’s not being recognised. Possibly one of your above edits to the user settings then is disrupting the macro & the system.

Graf911, 2024-12-18:
I downloaded the file and uploaded it unchanged and lo and behold everything started

but now, when running macros, the printer shows a message 
- Error evaluating 'gcode_macro _SAVE:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _DRIVER_VARS'
tell me what else I could have broken with my crooked handles

3DPrintDemon, 2024-12-18:
Ok great that fixed that one!

As for the new error, thats basically telling you that the SAVE macro can't find thing it's looking for, in this case its another macro called 'gcode_macro _DRIVER_VARS'.

This means the 'gcode_macro _DRIVER_VARS' is not in the system. 

So this tells us that the file `demon_homing_control_v1.4.0.cfg` has either been modified so it no longer works or it is not included in the system. This can be quite dangerous if your printer relies on the commands that are in this file!

Please check this file is unmodified, undamaged, in the correct directory & that you have used the correct include command for the Demon_Klipper_Essentials_Unified directory.

```
[include ./Demon_Klipper_Essentials_Unified/*.cfg]
```

This will automatically include all .cfg file inside the Demon_Klipper_Essentials_Unified directory.

Graf911, 2024-12-23:
Hi,
the printer has turned on and is even printing
but the printer reports an error
Unknown command:"SET_PIN"
Unknown command:"SET_FAN_SPEED"
but it prints normally.

at the end of the print, the message appears:
Error evaluating 'gcode_macro _DEMON_IDLE_TIMEOUT:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'fan_generic Nevermore'

help me solve this problem.

3DPrintDemon, 2024-12-23:
Hi sounds like you have turned on the Nevermore hardware option but either don’t have one or it’s not setup correctly.

Graf911, 2024-12-23:
the option is set
the configuration file defines
printer.cfg

#####################################################################
#                  FAN
#####################################################################

[fan]                                    # Parts Cooling Fan 
pin: EBBCan:gpio13                       # Pin for general fan control
kick_start_time: 0.5                     # Time to start the fan
off_below: 0.10                          # Temperature below which the fan turns off

#--------------------------------------------------------------------

[heater_fan hotend_fan]                  # hotend fan
pin: EBBCan:gpio14                       # Pin for hotend fan control
heater: extruder                         # Heater associated with this fan
heater_temp: 50.0                        # Temperature at which the fan activates

#--------------------------------------------------------------------

[heater_fan Nevermore]                   # Nevermore V6
pin: PA8                                 # Pin for Nevermore fan control
max_power: 1.0                           # Maximum power for the fan
shutdown_speed: 1.0                      # Speed at which the fan shuts down
kick_start_time: 5.0                     # Start-up time for the fan
heater: extruder                         # Heater associated with this fan
heater_temp: 200                         # Temperature at which the fan activates
fan_speed: 1.0                           # Fan speed

#--------------------------------------------------------------------

[heater_fan controller_fan]              # Skirt fan 1
pin: PE5                                 # Pin for Skirt fan 1
shutdown_speed: 0.0                      # Speed at which the fan shuts down (do not change)
kick_start_time: 0.5                     # Start-up time for the fan (do not change)
heater: heater_bed                       # Heater associated with this fan
heater_temp: 20                          # Temperature at which the fan activates
fan_speed: 0.7                           # Fan speed

#--------------------------------------------------------------------

[heater_fan controller_fan2]             # Skirt fan 2
pin: PD12                                # Pin for Skirt fan 2
shutdown_speed: 0.0                      # Speed at which the fan shuts down (do not change)
kick_start_time: 0.5                     # Start-up time for the fan (do not change)
heater: heater_bed                       # Heater associated with this fan
heater_temp: 50                          # Temperature at which the fan activates
fan_speed: 0.7                           # Fan speed

#--------------------------------------------------------------------

[heater_fan chamber]
pin: PD13                                
shutdown_speed: 0.0                      
kick_start_time: 0.5
heater: heater_bed
heater_temp: 65
fan_speed: 0.8

Graf911, 2024-12-23:
can I get an example of how you are configured?

closed 2024-12-30 by 3DPrintDemon