3DPrintDemon/Demon_Klipper_Essentials_Unified #75: My first update via moonraker

patofoto, 2024-12-19:
Getting this:

[image]

File disappeared

3DPrintDemon, 2024-12-19:
Hi. That’s a bit strange, it could be due to you using a custom named user account maybe. It’s normally best to use pi, or biqu for example depending on what system you have. 

The moonraker update manager system is not controlled by the macros but instead is under direct control of Klipper/moonraker. It will modify files of specified directories for its update process. The macros have no say or control over any of the update manager’s processes. 

It will update from & to where you tell it. 

I have tested this on 2 different machines here multiple times with no issues so far. 

Are you able to “ls” the directory to confirm the file is gone? Might be worth renaming your user account back to a standard name to see if it improves things.

patofoto, 2024-12-19:
Yes. Confirmed it is not there. Should I manually replace it from a backup?

patofoto, 2024-12-19:
uploaded a saved version of the `non_blocking_wait.cfg` file and I am back in business. Let me know if I can help you with anything to making sorting this out easily. I have never left my Raspberry Pi's main user stay as `pi` to make them more secure. I use them out in the wild as camera controllers and got in the habit of setting my own user. Let me know if I should revert it back to pi if that would be the solution but I imagine I am not the only one that changes the default user.

3DPrintDemon, 2024-12-19:
Ok great you’re back in business there, just keep an eye on the system & keep good backups just incase your instance decides it wants to more weird stuff for whatever reason. 
I mean the custom name is probably perfectly fine, I was just going by what I could see there in the image, after all it would be considered a variable in this case.

patofoto, 2024-12-19:
Copy. Will keep an eye out for weird things. Always try to have backups.

3DPrintDemon, 2024-12-19:
Cool! Definitely is! Also keep in mind SD card do eventually wear out & fail on these system.

3DPrintDemon, 2024-12-19:
I just ran an update manager update on one of my machines here that hadn’t been on since yesterday. All is fine, no errors & no missing files here atm.

3DPrintDemon, 2024-12-19:
ahh ok I see, won't be impending media failure most likely then. With any luck

closed 2024-12-23 by 3DPrintDemon