3DPrintDemon/Demon_Klipper_Essentials_Unified #77: Macros and Variables missing

javilopezalarcon, 2024-12-21:
Good morning, I'm seeing several issues that all point to the same thing and last night it happened to me.

I updated from moonraker and everything stopped working... I went to the docu and installed again with the .sh script but it didn't work either...

After analyzing the github, since no macro worked, no variable either, it turns out that the files were not there!!

I have seen that you changed their location, now they are in the demon_user_files_SOURCE folder, but that folder did not come in my installation, it is not there... I had to download it and place it in another location and add it to the printer.cfg so that it returns everything works as before...

I mention this because at least 2 or 3 issues before, it happened to me too, that gcode was missing: in the start or the errors... of course, there are no macros or variables, there is only our file where we renamed them to put them the good value.

I hope it helps you and sorry for my English, I use a translator.

I love your macros, greetings!!

3DPrintDemon, 2024-12-21:
Hi there. The file structure has changed as explained in [this section](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#new-feature-demon-user-files-updater).  

It talks about the new system & how the files need to place outside the main directory so they’re editable & wont spoilt or “dirty” the main install in the update manager & how you need to add a new include command & install the kiauh gcode shell expansion if you want the macros to do it for you. 

Or if not you have do [what is said here](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/main/Documentation/INSTALL_INSTRUCTIONS/General%20_Setup_For_All_Printers/INSTALL_INSTRUCTIONS.md#manual-user-file-placement) to copy (duplicate) the files out of the main install directory & include them in a new directory called `Demon_User_Files`

I hope that helps.

javilopezalarcon, 2024-12-21:
Well, I had not seen that change in the docu to update the macros, thank you very much, yes, I had taken the middle path and brought the files with me...

I have deleted them and used the installer adding the rpi include, but it goes into a loop, it never ends... as I read, it should bring some editable files but they do not appear... (although I have my own where I call the macro or variable and edit the value) I have recreated the folder and the files and so the extract loop no longer does it.

If there is anything you want me to try, let me know.

Thanks for the notice

3DPrintDemon, 2024-12-21:
There be a couple reasons for this, it’s most likely your variables file is incomplete, or possibly the files you added are of a different version so the extract prompt keeps telling you that they need to be updated. 
Also when new files are created you may need to reload you browser window to refresh the directory (as mentioned in the documentation & the prompt dialog). 

Make sure you have this in your printer.cfg…
```
[save_variables]
filename: ~/demon_vars.cfg
```
To refresh your variables file ssh this command…
```
cp -f ~/printer_data/config/Demon_Klipper_Essentials_Unified/Other_Files/Demon_Vars/demon_vars.cfg ~/
```

It’ll reset your system to think it’s the first time you boot up after install but your variables file will be complete.

javilopezalarcon, 2024-12-21:
Perfect, thank you very much.

I am going to use "plan b" since I have my macros.cfg and variables.cfg file where I have all the data saved and it worked perfectly, only now when updating I was missing the user files that I will put them on hand as you indicate in the docu in the second section.

Thank you very much for your prompt response, we will continue to enjoy your macros, greetings.

closed 2024-12-21 by javilopezalarcon

3DPrintDemon, 2024-12-21:
Fantastic I’m glad you found a solution! Happy printing! 😎👍