3DPrintDemon/Demon_Klipper_Essentials_Unified #78: Retraction calibration

subparstar, 2024-12-22:
I tried to use the recommended website for retraction calibration and the bed heats up, then the nozzle heats to printing temp, then it tries to home so it goes to 150c on the hotend. then error, extruder not hot enough

3DPrintDemon, 2024-12-22:
Hi. There are no bed or nozzle temperature modifiers in the _RET_CALI_START macro or any or the macros it calls. 

Please be sure you’re putting _RET_CALI_START only into the retraction calibration website start Gcode as per the instructions.

subparstar, 2024-12-22:
I just tried it again. Bed heats up to 110c, then nozzle heats to 260c, then it homes x and y, it attempts to home z but the temp is too high at 260c so it lowers the temp to 150c, it then homes z when it reaches 150c, then cancels the print because extruder below minimum temp. I added my klippy log if that helps
[klippy (1).log](https://github.com/user-attachments/files/18222059/klippy.1.log)

3DPrintDemon, 2024-12-22:
I can see in your log you're using Voron Tap. This is the problem, as Tap requires the nozzle to be at 150º maximum for Homing Z, it is not a fault with the macro behaviour but an override from Tap. 

https://github.com/VoronDesign/Voron-Tap/blob/main/config/tap_klipper_instructions.md

As these macros are not optimised for Tap you can edit the _RET_CALI_START if you need the function immediately. If you can wait I probably could bring in something for a future update.

Find the original macro to modify [here](https://github.com/3DPrintDemon/Demon_Klipper_Essentials_Unified/blob/e8f58ee569d86378886b5c6bac20afe1fa297514/demon_print_start_end_v2.9.4.cfg#L566)

Then add in this after the G28 Z homing events.
```
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET=260
  TEMPERATURE_WAIT SENSOR=extruder MINIMUM=260
```
In the website's start gcode box you could then set your nozzle temperature to say 180 on the website. The command you added should then raise the temp again after Z homing.

That should sort out your issue. Please remember that the initial retraction website's starting temperature as well as the Tap temperature overrides when homing Z are out of control of the Demon macro set currently.

subparstar, 2024-12-24:
I tried your temporary solution and could not get it to work correctly. I will wait to see if you're able to get TAP to function with this macro. I enjoy everything else about this. I can tell alot of hard work went into it. Happy holidays!

3DPrintDemon, 2024-12-27:
Ok ill try to take a look at a solution when I have some time.