3DPrintDemon/Demon_Klipper_Essentials_Unified #81: homing_override and safe_z_homing cannot be used simultaneously

youziStar, 2024-12-27:
After my content includes the following:
[image]
I used Cartographer Probe，I suspect it's the problem。
Below is my print configuration, everything else should be default.
[printer.txt](https://github.com/user-attachments/files/18262337/printer.txt)

3DPrintDemon, 2024-12-27:
Yes the problem is you have safe homing & homing override enabled. The macros have their own homing override system, so you’ll need to disable your safe homing section.

Tofusang, 2024-12-28:
I m using IDM as well, comment out your safe_z_home section will do the job. But I have questions of how to better integrate IDM with DemonKlipper macros.

closed 2025-01-05 by 3DPrintDemon