3DPrintDemon/Demon_Klipper_Essentials_Unified #84: Random Cleaning Nozzle Error

Tofusang, 2024-12-28:
Voron2.4 R2 300 brush&purge bucket on rear left hand rail.
When I hit Clean Nozzle, i got this error msg:
Error evaluating 'gcode_macro _step_control:gcode': jinja2.exceptions.UndefinedError: No random item, sequence was empty.
I have to turn off below section to use linear cleaning to make things happen
[image]
Any thing that I missed to be able to use the random cleaning function?
Attached is my user setting . 

[demon_user_settings_cleaner_variables_v1.1.0.txt](https://github.com/user-attachments/files/18267885/demon_user_settings_cleaner_variables_v1.1.0.txt)
[demon_user_settings_v2.9.4.txt](https://github.com/user-attachments/files/18267887/demon_user_settings_v2.9.4.txt)

3DPrintDemon, 2024-12-28:
Hi there, I just had a quick look & the only issue I can see immediately in your file is that you have the min/max X cleaning values reversed in the `demon_user_settings_cleaner_variables_v1.1.0.cfg`
```
YOUR VALUES...
variable_clean_min_x: 126
variable_clean_max_x: 88 
```
The minimum is the lower number on the axis & max is the higher. The cleaning should start at your... 

```
variable_start_x: 126 
```
then move to random points within the corrected range below.

```
CORRECTED VALUES...
variable_clean_min_x: 88
variable_clean_max_x: 126 
```

Obviously please increase you're cleaning Z height to say 10-15mm to test.

Tofusang, 2024-12-28:
Thanks mate! This definitely will do. Great work on this project!

closed 2024-12-28 by Tofusang

3DPrintDemon, 2024-12-28:
You're most welcome! Im glad you're enjoying my work.